jsk-ros-pkg/jsk_pr2eus #103: problem of "generate-all-msg-srv.sh"

chiwunau, 2015-02-13:
I just updated to the latest roseus and installed ros-hydro-euslisp and ros-hydro-jskeus,
when I try to run pr2-interface.l the below errors come out 
```
load "package://pr2eus/pr2-interface.l"
[ERROR] [1423838935.993902922]: Could not find roseus messages for /opt/ros/hydro/share/pr2_msgs unde\
r (/home/au/ros/hydro/devel/share/roseus/ros /opt/ros/hydro/share/roseus/ros)                         
try rosrun roseus generate-all-msg-srv.sh /opt/ros/hydro/share/pr2_msgs
[ERROR] [1423838935.993957259]: Could not find roseus messages for /opt/ros/hydro/share/pr2_msgs unde\
r (/home/au/ros/hydro/devel/share/roseus/ros /opt/ros/hydro/share/roseus/ros)                         
try rosrun roseus generate-all-msg-srv.sh /opt/ros/hydro/share/pr2_msgs
[ERROR] [1423838935.994930256]: Could not find roseus messages for /opt/ros/hydro/share/pr2_controlle\
rs_msgs under (/home/au/ros/hydro/devel/share/roseus/ros /opt/ros/hydro/share/roseus/ros)             
try rosrun roseus generate-all-msg-srv.sh /opt/ros/hydro/share/pr2_controllers_msgs
[ERROR] [1423838935.994965207]: Could not find roseus messages for /opt/ros/hydro/share/pr2_controlle\
rs_msgs under (/home/au/ros/hydro/devel/share/roseus/ros /opt/ros/hydro/share/roseus/ros)             
try rosrun roseus generate-all-msg-srv.sh /opt/ros/hydro/share/pr2_controllers_msgs
/opt/ros/hydro/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: no such package PR2_CONTROLLERS\
_MSGS in (apply #'ros::load-org-for-ros ros::fullname args)
```

```
rosrun roseus generate-all-msg-srv.sh /opt/ros/hydro/share/pr2_msgs
[rospack] Error: stack/package /opt/ros/hydro/share/pr2_msgs not found
generating... 0 files with ALL=No, COMPILE=No option
0
```

It seem like something related to the package path, anyone faced the same problem?

k-okada, 2015-02-13:
try rosrun roseus generate-all-msg-srv.sh pr2_msgs

chiwunau, 2015-02-13:
rosrun roseus generate-all-msg-srv.sh pr2_msgs passed, 
but the same error still occur.

k-okada, 2015-02-13:
Please paste
- what did you input 
- what was the output, past whole output
If the data is huge , check gist service

orikuma, 2015-02-13:
Can this change solve the problem?
(I don't know this workaround is correct)

diff --git a/src/geneus/geneus_main.py b/src/geneus/geneus_main.py
index 0680c49..ede32e5 100644
--- a/src/geneus/geneus_main.py
+++ b/src/geneus/geneus_main.py
@@ -72,7 +72,7 @@ def genmain(argv, progname):
             f.write(";; DO NOT EDIT THIS FILE\n")
             f.write(";;\n")
             for pkg in args[1:]:
-                f.write("(ros::load-ros-package \"%s\")\n"%pkg)
+                f.write("(ros::load-ros-package
\"%s\")\n"%os.path.basename(pkg))
             f.close()
             retcode = 0
         else:

k-okada, 2015-02-13:
i see, fixed in https://github.com/jsk-ros-pkg/jsk_roseus/pull/233

chiwunau, 2015-02-13:
it works now by @k-okada's PR
thank you.

closed 2015-02-13 by chiwunau