jsk-ros-pkg/jsk_pr2eus #120: send angle-vector to pr2 too slow

furushchev, 2015-03-18:
when using torso, moving arms are slow. I think this is because max joint vector of torso controller is too much slower than other controllers.

e.g.:

```lisp
irteusgl$ setq a (make-coords :pos #f(700 0 1300))
irteusgl$ send *pr2* :inverse-kinematics a :use-torso t
#f(242.932 60.0 74.0 70.0 -120.0 20.0 -30.0 180.0 3.13996 0.865655 -29.5923 -76.9422 -183.126 -77.7605 210.538 0.0 0.0)
irteusgl$ bench (send *ri* :angle-vector (send *pr2* :angle-vector) 2000)
irteusgl$ bench (send *ri* :angle-vector c 2000) ;; this assumes 2.0s to move body
;; time -> 25.3008[s] ;; this takes too long
#f(242.932 60.0 74.0 70.0 -120.0 20.0 -30.0 180.0 3.13996 0.865655 -29.5923 -76.9422 -183.126 -77.7605 210.538 0.0 0.0)
```

Any idea?

my idea is porting current pr2-interface's `:angle-vector` to `:angle-vector-safe` and implement `:angle-vector` like this:

```lisp 
 (:angle-vector
   (av &optional (tm 3000) &rest args)
     (send self :angle-vector-safe av tm (remove :torso-controller controller-type))))
```

garaemon, 2015-03-18:
scale optional argument is too large. we should use scale 1 to check if the
duration is too fast

k-okada, 2015-03-18:
Are you saying that;
even if you send `(send *ri* :angle-vector c 2000)`, the pr2
controller system moves very slowly and it takes 25 sec,
OR
when you call `(send *ri* :angle-vector c 2000)`, the :angle-vector
method changes tm value from 2000 to 25000 and publish to actionlib of
pr2 controller,

In the latter case, we have to check robot-interface/pr2-interface
code, in the former case, we need to check ik code.
If I remember correctly, I wrote a code not to use torso too much when
we sokve ik, that's should be somewhere in pr2?
--
◉ Kei Okada

furushchev, 2015-03-18:
I mean the latter case, I changed :scale to 1 (default is 5).

```lisp
$ bench (send *ri* :angle-vector c 2000 :default-controller 0 :scale 1)
;; time -> 5.13268[s] ;; 5 times faster, but still is not 2000ms
#f(242.932 60.0 74.0 70.0 -120.0 20.0 -30.0 180.0 3.13996 0.865655 -29.5923 -76.9422 -183.126 -77.7605 210.538 0.0 0.0)
```

This result might be more reasonable for me. 
(it means we used to move torso faster than its joint velocity limit)

closed 2015-03-19 by garaemon