jsk-ros-pkg/jsk_pr2eus #149: default の pr2_descriptionへの変更

k-okada, 2015-07-07:
https://github.com/jsk-ros-pkg/jsk_robot/issues/366
https://github.com/jsk-ros-pkg/jsk_demos/issues/987#issuecomment-117890195

ですが残りの作業としてpr2.lを変えないといけないというのがあると思います．正しいですか？

- [ ] bag ファイルを撮り直す　（https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/test/pr2eus-test.launch#L3) 
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/test/pr1012_camera_info.bag
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/test/pr1012_urdf.yaml
- [ ] pr2.l　生成プログラムのフレーム周りを更新
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/make-pr2-model-file.l#L47
- [ ] pr2.l ファイルを生成
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/test/make-pr2-model-file-test.launch

furushchev, 2015-07-08:
@k-okada 合っていると思います。
ひと通りやって見ましたが、最後で以下のエラーが出ました。

```
configuring by "/home/furushchev/ros/hydro_parent/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x4896420[16374] --> 0x4ccbb00[32748] top=37b8
irtgl irtviewer 
EusLisp 9.12(fada0ef 1.0.4) for Linux64 created on pr1012(Thu Jun 18 07:20:07 PDT 2015)
roseus ;; loading roseus("6c13302") on euslisp((9.12 pr1012 Thu Jun 18 07:20:07 PDT 2015 fada0ef 1.0.4))
eustf roseus_c_util ;; WARNING link without geometry nor node: base_laser_link geometry : 0, node : 0
;; WARNING link without geometry nor node: head_chain_cb_link geometry : 0, node : 0
;; WARNING link without geometry nor node: head_mount_kinect_ir_optical_frame geometry : 0, node : 0
;; WARNING link without geometry nor node: head_mount_kinect_rgb_optical_frame geometry : 0, node : 0
;; WARNING link without geometry nor node: head_mount_prosilica_optical_frame geometry : 0, node : 0
;; WARNING link without geometry nor node: projector_wg6802418_child_frame geometry : 0, node : 0
;; WARNING link without geometry : projector_wg6802418_frame
;; WARNING link without geometry nor node: narrow_stereo_l_stereo_camera_optical_frame geometry : 0, node : 0
;; WARNING link without geometry nor node: narrow_stereo_r_stereo_camera_optical_frame geometry : 0, node : 0
;; WARNING link without geometry : narrow_stereo_r_stereo_camera_frame
;; WARNING link without geometry : narrow_stereo_l_stereo_camera_frame
;; WARNING link without geometry nor node: narrow_stereo_optical_frame geometry : 0, node : 0
;; WARNING link without geometry : narrow_stereo_link
;; WARNING link without geometry nor node: wide_stereo_l_stereo_camera_optical_frame geometry : 0, node : 0
;; WARNING link without geometry nor node: wide_stereo_r_stereo_camera_optical_frame geometry : 0, node : 0
;; WARNING link without geometry : wide_stereo_r_stereo_camera_frame
;; WARNING link without geometry : wide_stereo_l_stereo_camera_frame
;; WARNING link without geometry nor node: wide_stereo_optical_frame geometry : 0, node : 0
;; WARNING link without geometry : wide_stereo_link
;; WARNING link without geometry nor node: high_def_optical_frame geometry : 0, node : 0
;; WARNING link without geometry : high_def_frame
;; WARNING link without geometry nor node: imu_link geometry : 0, node : 0
;; WARNING link without geometry nor node: l_forearm_cam_optical_frame geometry : 0, node : 0
;; WARNING link without geometry : l_forearm_cam_frame
;; WARNING link without geometry nor node: l_gripper_led_frame geometry : 0, node : 0
;; WARNING link without geometry nor node: l_gripper_motor_screw_link geometry : 0, node : 0
;; WARNING link without geometry : l_gripper_motor_slider_link
;; WARNING link without geometry nor node: l_gripper_l_finger_tip_frame geometry : 0, node : 0
;; WARNING link without geometry nor node: l_gripper_tool_frame geometry : 0, node : 0
;; WARNING link without geometry nor node: left_arm_chain_cb_link geometry : 0, node : 0
;; WARNING link without geometry nor node: l_torso_lift_side_plate_link geometry : 0, node : 0
;; WARNING link without geometry nor node: laser_tilt_link geometry : 0, node : 0
;; WARNING link without geometry nor node: r_forearm_cam_optical_frame geometry : 0, node : 0
;; WARNING link without geometry : r_forearm_cam_frame
;; WARNING link without geometry nor node: r_gripper_led_frame geometry : 0, node : 0
;; WARNING link without geometry nor node: r_gripper_motor_screw_link geometry : 0, node : 0
;; WARNING link without geometry : r_gripper_motor_slider_link
;; WARNING link without geometry nor node: r_gripper_l_finger_tip_frame geometry : 0, node : 0
;; WARNING link without geometry nor node: r_gripper_tool_frame geometry : 0, node : 0
;; WARNING link without geometry nor node: right_arm_chain_cb_link geometry : 0, node : 0
;; WARNING link without geometry nor node: r_torso_lift_side_plate_link geometry : 0, node : 0
;; WARNING link without geometry nor node: torso_lift_motor_screw_link geometry : 0, node : 0
load model file from parameter server /robot_description to /tmp/pr2_3082.l
received narrow_stereo/left/camera_info #<sensor_msgs::camerainfo #X49ebe40> narrow_stereo_optical_frame
received narrow_stereo/right/camera_info #<sensor_msgs::camerainfo #X4b2b648> narrow_stereo_optical_frame
received wide_stereo/left/camera_info #<sensor_msgs::camerainfo #X4b29db8> wide_stereo_optical_frame
[... cut ...]
[ROSTEST]-----------------------------------------------------------------------

[testmake_pr2_model_file_test][failed]

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 0
 * FAILURES: 1

ERROR: The following tests failed to run:
 * testmake_pr2_model_file_test

rostest log file is in /home/furushchev/.ros/log/rostest-pr1012-2965.log
```

bagファイルの中で、`/kinect_head/depth/camera_info`が出ていないのが原因と思ったのですが、そもそもこのトピックは1回もpublishされない気がしています。
(pr1012の`/kinect_head/depth/camera_info`もpr1040(jskオリジナルurdfの方)の`/openni/depth/camera_info`も`robot start`してから1回も出ていないです。)
これはそういうものなのでしょうか？

k-okada, 2015-07-08:
camera_info がでていないことはないともいます．openni.launch単体で立ち上げて試してみてください．

furushchev, 2015-07-27:
@k-okada 対応が遅れました。 rqt_reconfigureでdriverの`depth_registration`をオフにしたところ、`/kinect_head/depth/camera_info`からパブリッシュされるようになりました。

furushchev, 2015-09-10:
上記のPRで対応できているつもりです。
冷蔵庫でもとエレベータは実機で動くことを確認しました。

k-okada, 2015-09-12:
まだ，これは1012だけだっけ？であれば1040も対応しましょう．

closed 2015-09-12 by k-okada

furushchev, 2015-09-12:
ｐｒ１０４０も対応済みです。