jsk-ros-pkg/jsk_pr2eus #150: error in robot start command

h-kamada, 2015-07-14:
When using
``` roslaunch /etc/ros/hydro/robot.launch ```
launch is executed.
But when using
``` robot start ```
robot start fails with following error.

h-kamada, 2015-07-14:
solved.
/var/tmp/ros.pid  must be removed when using 
``` robot start```
command.
Not knowing why.

garaemon, 2015-07-14:
have you run robot claim?

furushchev, 2015-07-14:
I think this sometimes occurs these days only on users without applications on PR1012.

```
su - pr2admin
sudo rm /var/tmp/ros.pid
```

by executing above, it's clear.
Weirdly, permisstions and owner to /var/tmp/ros.pid are all the same as on PR1040.
All users on PR1040 and applications user on PR1012 seems ok.

garaemon, 2015-07-14:
check user group

furushchev, 2015-07-18:
sorry for late response.

user/group of `/var/tmp/ros.pid` are both the user which executes `robot start` with permission `-rw-rw-r--`
Groups which users are belonged in are

```bash
pr1012 $ id -a
uid=1009(furushchev) gid=1010(furushchev) groups=1010(furushchev),100(users),998(apt)
```

```bash
pr1040 $ id -a
uid=1008(furushchev) gid=1009(furushchev) groups=1009(furushchev),100(users),998(apt)
```

also same.

furushchev, 2015-08-10:
It seems that, the problem is permission `-rw-rw-r--`.
`roslaunch` does not remove and re-create `ros.pid`, but rewrites it.
I added write permission for all users, and it works now.
`pr2admin $ sudo chmod a+w /var/tmp/ros.pid`

furushchev, 2015-11-28:
`robot stop` usually sends `SIGINT` to roslaunch which creates `/var/tmp/ros.pid`, and escalates up to `SIGKILL` if killing roslaunch seems not work well. (see https://github.com/furushchev/pr2/blob/patch-1/pr2-core/root/usr/lib/robot/robot.py#L604 )
If signal is above `SIGINT`, callback on shutdown of roslauch is not called. (this lets `/var/tmp/ros.pid` remain without being removed)
And `/var/tmp/ros.pid` is created as linux default permission 0664 (cf. umask ), which other users cannnot remove without `sudo`.
Possible solution is to change `umask` of users from 0002 to 0000, but it may be not so good in security reason...

```bash
# append to .bashrc of each pr2 users
umask 0000
```

furushchev, 2015-11-28:
I created PR, tested and patched on pr1012.
It should solve this issue.

furushchev, 2015-12-04:
please close if you do not have this problem now.

closed 2015-12-04 by h-kamada