jsk-ros-pkg/jsk_pr2eus #160: pr2の:go-posはブロックする？

k-okada, 2015-08-06:
```
:go-pos
```
するとゴールにつくまで帰ってこない？帰ってくる？
hrp2は
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/euslisp/rtm-ros-robot-interface.l#L716

:go-pos, :go-pos-no-wait, :wait-foot-step
があるみたいだけど，同様なのはないんだっけ？

動きながら，手を動かす，みたいなコードはどうやって書いている？ @h-kamada

h-kamada, 2015-08-07:
go-posは返ってこないものだと思ってます。(後で調べます)
少なくともgo-pos-unsafeは返ってきません。
卒論の時に書いたコードでは、手とbaseの動きがしっかり連動して欲しかったのでgo-pos-unsafeを用いていましたが、これは終わるまで返ってこないので、関数内でwaitしている部分だけ除いて使っていました。
hrp2でもそのような関数があるのならばPR2でもそのような関数を足しても良いでしょうか。

k-okada, 2015-08-07:
なるほど，関数は
:go-pos
:go-pos-no-wait
:go-wait (go-pos をwaitする）
:go-stop (止まる．これはもうある）
にしましょうか．

作っとくので，それを前提にデモ関数を変えていって下さい．

ところで
冷蔵庫デモをeuslispだけでやるのはどうするの？
https://github.com/jsk-ros-pkg/jsk_demos/tree/master/detect_cans_in_fridge_201202
? Readme つくっておいて，

https://github.com/jsk-ros-pkg/jsk_demos みたいに，
またこのjsk_demos/README.mdは
https://github.com/jsk-ros-pkg/jsk_demos/tree/master/jsk_2013_04_pr2_610
に統合して，トップページはカッコいい写真とリンクの羅列にしようか．

h-kamada, 2015-08-07:
> なるほど，関数は :go-pos :go-pos-no-wait :go-wait (go-pos をwaitする） :go-stop (止まる．これはもうある） にしましょうか．
:go-pos と:go-waitの違いは何でしょうか。

また、go-pos-unsafeのwait有り無しバージョンは作らないのでしょうか？

k-okada, 2015-08-07:
go-pos = go-pos-no-wait + go-wait

にします．go-wait はgoと付いていますが，進まず待つだけです．go-stopが止まるだけなのように．

go-pos-unsafeもつくってもいいですが，基本は全てgo-posで動くはずです．（うまくコストマップを替える等々）

k-okada, 2015-08-07:
> 冷蔵庫デモをeuslispだけでやるのはどうするの？
> https://github.com/jsk-ros-pkg/jsk_demos/tree/master/detect_cans_in_fridge_201202 ? Readme つくっておいて，

以下で必要 https://github.com/himlen1990/todo/issues/1

h-kamada, 2015-08-07:
確認しましたが、go-pos も終了するまで返って来ません。

k-okada, 2015-08-07:
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/162

で，go-pos, go-pos-no-wait を足しました．一発で動く気がしないので，書くにして下さい．

あと，元のコードからですが，
map-to-frame が解決できた時と出来ないときで，wait
のwhen ret いかでちゃんと対応出来ていない気がします．

h-kamada, 2015-08-07:
> あと，元のコードからですが， map-to-frame が解決できた時と出来ないときで，wait のwhen ret いかでちゃんと対応出来ていない気がします．

これは#157 であげられていた問題でしょうか？
> [ERROR] [1438838240.968802675]: "map" passed to lookupTransform argument
> target_frame does not exist
> ```
> message, but everything working well, is it correct? if so please write
> issue for this. Displaying  error, but works good is confusing.

もしそうであれば jsk-ros-pkg/jsk_demos#1026 のPRで解決できたと思っているのですが...

k-okada, 2015-08-07:
ちがいます．
また，上のerrorは
             (map-to-frame (send *tfl* :lookup-transform "/map" frame-id
(ros::time 0)))
ででるけど，simulation-modep の時もこのコードにはいって，でも使わない，というのが問題では

k-okada, 2015-10-05:
Cf: https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/163

closed 2015-10-05 by k-okada