jsk-ros-pkg/jsk_pr2eus #171: :go-posなどが失敗したらnilを返す,

furushchev, 2015-09-12:


furushchev, 2015-09-12:
ref https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/164

k-okada, 2015-09-12:
なるほど．これって実機/gazebo無しのkinematics simulator 環境でテストかけるのかな？
とりあえず
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/163/files
の説明(https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/163/files#diff-15f6920b924513f6027d36b361f85bbdR826)のところに ，希望する振る舞いを書いてみましょう．

Cc: @snozawa

snozawa, 2015-09-12:
実はこのあたりがrobot-interface.lにはいってなくて
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/163
で追加になるので今までよくわかってなかったのですが、
go-posが失敗するというのはどいうった状況のことでしょうか。

またこれに限らず、指令値送ることに関係するメソッド（:angle-vector, angle-vector-sequence, go-pos, :wait-interpolation...）は
今まで基本的には返り値をとってませんでしたが、これらも同様に
うまくいってたらnil以外、なんらかの問題があればnilをかえす、というのでよいようにも思います。
（もしくは、うまくいったらtで失敗したらエラーフラグに対応したシンボルをかえす）

k-okada, 2015-09-12:
pr2のgo-posはmobe_base経由だから、障害物があればそこでとまってゴールまで到達しないのです