jsk-ros-pkg/jsk_pr2eus #178: pr2 Kinematics Simulatorでロボットにobjectを:assoc/:dissocしたい

furushchev, 2015-11-02:
[image]

pr2 kinematics simulatorで物体の把持などを可視化したい時はどうしたらいいでしょうか？

k-okada, 2015-11-02:
send *ri* :robot :assoc (elt (send *ri* :objects) 0)
かな
send *ri* :find-object
とかあってもいいかもしれない。
https://github.com/euslisp/jskeus/blob/master/irteus/irtscene.l みたいに

furushchev, 2015-11-02:
:assocする対象は1つなので、どちらのrobotにassocさせるかという問題がある木がします。
例えば

```
(send (send (send *pr2* :rarm :end-coords) :parent) :assoc *cup*)
(send (send (send *ri* :robot :rarm :end-coords) :parent) :assoc *cup*)
```

とすると、最初の`*pr2*`にassocされるため、`(send *ri* :robot)`へのassocは失敗(無視)します。

```
(send *ri* :objects (list *cup*)
```

したときにメソッド内部でcopy-objectするというのが考えられますが、これは元のobjectをtransformしたときにずれて行ってしまうので、微妙な気もします。。。

furushchev, 2015-11-03:
robot-interfaceの`:objects`メソッド内でcopy-objectしているようなので、これで目的は達成できそうです。
ありがとうございました。

closed 2015-11-03 by furushchev