jsk-ros-pkg/jsk_pr2eus #181: pr2-interfaceでgripperの現在のposition(グリッパー間が何m(or mm)か？)を知りたい

furushchev, 2015-11-03:
現在実機では

```
4.-irteusgl$ send *ri* :start-grasp :rarm
;; #<rotational-joint #Xa1bfa00 r_elbow_flex_joint> :joint-angle(-121.624) violate min-angle(-121.542)
1.5608
```

のように`:start-grasp`だけ返り値がpositionになりますが、

- `start-grasp`のようにアクチュエータを動かすことなくpositionが知りたい
- pr2-robot (eusモデル) でも知りたい

場合はすでに何か方法があるでしょうか？

k-okada, 2015-11-12:
なんにもならないコメントですが，指間の距離ってかpublishされていなかったっけ？

YoheiKakiuchi, 2015-11-21:
> なんにもならないコメントですが，指間の距離ってかpublishされていなかったっけ？

gripperのactionのフィードバックかstate？かに出てないですかね？

あとは、joint_statesを見るんでしょうか。

k-okada, 2015-11-30:
https://github.com/PR2/pr2_controllers/blob/hydro-devel/pr2_controllers_msgs/action/Pr2GripperCommand.action
で result/state で出ていそうですね

furushchev, 2015-12-16:
closed via #190

closed 2015-12-16 by furushchev