jsk-ros-pkg/jsk_pr2eus #187: :wait-interpolationの正しい挙動

furushchev, 2015-11-20:
:wait-interpolationについて、

1. robot-interface.lのdocstring
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L478
では
`values is a list of interpolatingp for all controllers, so (null (some #'identity (send *ri* :wait-interpolation))) -> t if all interpolation has stopped`
とあります。
これはつまりtimeoutが0(無限大)で:wait-interpolationされたときに必ず`(nil nil nil nil)`になるという想定
testコードは
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/test/default-ri-test.l#L29

2. 実機につないだ時のwait-interpolation
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L487
では`:wait-interpolation`の返り値は`(send-all controller-actions :wait-for-result :timeout timeout)`
となっています。
  - `values is a list of interpolatingp for all controllers`では無いように思います。
  - これは`:wait-for-result`はresultが返ってくると`t`を返す関数なので、actionがabortされない限り`t`を返すと思います。さらに1回でもactionを送ると、次からはtが返ります。(直前のgoalに対するwait-for-resultの結果)

pr2実機の場合

```bash
$ roscd pr2eus
$ roseus pr2-interface.l
$ pr2-init
$ send *ri* :wait-interpolation
[ INFO] [1448030250.007490532]: wait-interpolation debug: start
[ INFO] [1448030250.007857212]: wait-interpolation debug: end
(nil nil nil nil)
$ send *ri* :angle-vector (send *pr2* :reset-pose) 10000
#f(50.0 60.0 74.0 70.0 -120.0 20.0 -30.0 180.0 -60.0 74.0 -70.0 -120.0 -20.0 -30.0 180.0 0.0 0.0)
$ send *ri* :wait-interpolation
[ INFO] [1448030550.250459313]: wait-interpolation debug: start
[ INFO] [1448030552.680471962]: wait-interpolation debug: end
(t t t t)   ;; docstringと違う？
$ send *ri* :wait-interpolation
[ INFO] [1448030579.632002770]: wait-interpolation debug: start
[ INFO] [1448030579.632373273]: wait-interpolation debug: end
(t t t t)  ;; docstringと違う?
```

3. 実機なしのwait-interpolation
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L482
となっていて、docstringに従っている

simulationの場合

```bash
$ roscd pr2eus
$ roseus pr2-interface.l
$ pr2-init
$ send *ri* :wait-interpolation
(nil nil nil nil)
$ send *ri* :angle-vector (send *pr2* :reset-pose) 10000
#f(50.0 60.0 74.0 70.0 -120.0 20.0 -30.0 180.0 -60.0 74.0 -70.0 -120.0 -20.0 -30.0 180.0 0.0 0.0)
$ send *ri* :wait-interpolation
(nil nil nil nil)
$ send *ri* :wait-interpolation
(nil nil nil nil)
```

YoheiKakiuchi, 2015-11-21:
docstringに合わせるのがAPIとしてスッキリしていると思いますが、
実機でwait-interpolatinの結果を見て何かしているところに影響がありますね。
pr2/hrp2/urata系のすべてに影響が及びます。

それで、実機の場合のwait-interpolationの結果はwait-for-resultの結果ではなくて、
```send *ri* :interpolatingp```の結果を返すのがいいかと思います。
docstring通りですし、interpolatingpの結果は説明通りのように見えます。

k-okada, 2015-11-23:
おそくなりました．

うーん，いまいちですね．
ただ，:wait-interpolation でかえるのは，wait-for-result だとして，これがnil になるのは，
followtrajectoryactionでエラーが大きくなった時か，
https://github.com/start-jsk/rtmros_common/pull/765
がmergeされてて，canel させた時しか無いのではないでしょうか．

つまり，hrp2/urata系には影響なく，影響あるのはpr2だけかと．

ところで，pr2ではどの場面でこれがもんだいになりましたか．

furushchev, 2015-11-23:
> ところで，pr2ではどの場面でこれがもんだいになりましたか．

例えば

> https://github.com/jsk-ros-pkg/jsk_demos/blob/master/jsk_demo_common/euslisp/attention-observation.l#L29
> などではwait-interpolationが終わると`(t t t t)`を返すような想定になっています。

ではsimulationの場合wait-interpolation中に必ず中断が入る挙動になってしまいます。
jsk_demo_commonのテストコードを書こうとしておかしいなと気づきました。

k-okada, 2015-11-24:
なるほど．:wait-interplationの結果に依存するコードはないということでいいですか．

> 実機の場合のwait-interpolationの結果はwait-for-resultの結果ではなくて、
> send *ri* :interpolatingpの結果を返すのがいいかと思います。

でやりましょう．

snozawa, 2015-11-28:
こちらですが、pr2がやわらくてangle vectorをおおくってなにかにぶつかってとまり、そのときの間接エラーがおおきいとき、wait-interpolationがnilになったりするでしょうか。

k-okada, 2015-11-28:
はい．そう期待しています．
実際にどうなるか，明示的に確認したことはあまりない気がするので，一回試したらいいですね　＞ @furushchev @h-kamada @chiwunau @himlen1990

snozawa, 2015-11-28:
初歩的な質問になりますが，
- 実際はエラーはどのがプログラムどのようにして判定してしてるでしょうか．実関節角度値はjoint_statesでしょうか
- HrpsysROSBridge系であればどれがそれになるでしょうか．今も（正しく動いてたら）:wait-interpolationがHrpssyROSBridgeから返った時に判定結果がえられるのでしょうか．

furushchev, 2015-11-29:
> 実際にどうなるか，明示的に確認したことはあまりない気がするので，一回試したらいいですね

```
37.irteusgl$ (send *pr2* :rarm :move-end-pos #f(0 0 500) *pr2*)
#f(50.0 60.0 74.0 70.0 -120.0 20.0 -30.0 180.0 -12.4833 16.5145 -51.6378 -76.3784 58.4912 -23.4754 122.546 -24.5457\
 59.33)
38.irteusgl$ (progn (send *ri* :angle-vector (send *pr2* :angle-vector)) (send *ri* :wait-interpolation))
[ INFO] [1448786993.901075475]: wait-interpolation debug: start
[ INFO] [1448786997.010105078]: wait-interpolation debug: end
(t t t t)
```

などとして(アームの上方向300mmくらいに机がある環境で500mm上にend-coordsを動かしてみる。)試してみたところ、アームに関しては机と衝突して目標位置に行けなくても`t`が帰ってきているように思います。
返り値自体は各コントローラがresultを返したかどうかという意味での`t`でしょうか。
アームに司令を送り、他のノードからcancelを送るとnilが帰ってくるのは確認しています。

k-okada, 2015-11-29:
pr2/ros 系の実装は
https://github.com/ros-controls/ros_controllers/blob/4508d8797b245004c6aa3fa5722eb8e23a27208d/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L200
のあたりだともいます．
hrpsys ros bridgeではやっていないと思いますので，やるべきという声は有りますか？という話でした．
https://github.com/start-jsk/rtmros_common/pull/765
も，含めて，必要，ということがあれば，issueを抱いておいて下さい．

furushchev, 2015-11-30:
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/191 のように関数をdocumentに従うように修正して見ました。

k-okada, 2017-05-22:
ちなみに、`:wait-interpolation-smooth`の挙動はどうでしょうか？docstringは`Return value is a list of interpolatingp for all controllers`ですが、そうなっていないような気がします．

k-okada, 2017-05-23:
`:interpolating-smoothp`の挙動もdocstringと異なり常にnilでしょうか．

furushchev, 2017-05-23:
`:wait-interpolation-smooth`はPR2では使われていないのでわからないですが、`:wait-interpolation`と同じ場所で使われるなら揃えたほうがいいのではないでしょうか？
(CCでmentionしたらいい人もわからない。。。 @wkentaro ?)

furushchev, 2017-05-23:
（これテストしたいところですが、gazeboをテストに入れると途端にテストの速度低下＆＆エラー率が上がるので、何とかしたいところですね。。。）

knorth55, 2017-05-23:
@pazeshunがsmoothを使っています

pazeshun, 2017-05-23:
smoothの戻り値をチェックしたことがないので、わからないです。

pazeshun, 2018-07-14:
> ちなみに、:wait-interpolation-smoothの挙動はどうでしょうか？docstringはReturn value is a list of interpolatingp for all controllersですが、そうなっていないような気がします．

> :interpolating-smoothpの挙動もdocstringと異なり常にnilでしょうか．

大変遅くなりましたが、 #356 を作っていて、`:wait-interpolation-smooth`について色々確認しました。
@k-okada のご指摘通り、`:wait-interpolation-smooth`が終了した際には常にnilを返すようになっていましたので、 #356 で`:wait-interpolation`と同じくdocstringに従うようにしました。

なお、`(send *ri* :wait-interpolation-smooth 500)`と送った場合は、補間終了の500ms前に関数から抜けることになるので、返り値はtを含んだlistになると予想されますが、まだ実機で確認していません。

pazeshun, 2018-07-14:
#351 #352 がどうmergeされるかによって、`:wait-interpolation`の仕様が変わりそうですが、変わった場合は`:wait-interpolation-smooth`もそれに合わせていこうと思います。

furushchev, 2018-07-31:
放置していましたが，まとめとして，

docstringに書いてある

> return values is a list of interpolatingp for all controllers, so (null (some #'identity (send ri :wait-interpolation))) -> t if all interpolation has stopped

を原則として，

- `:wait-interpolation`はactionlibを想定して，`:angle-vector`などを使って，action serverに投げたgoalに対して，resultが返ってくるまで待つ．ということになる. (i.e. `:wait-interpolation`はすべてのresultを待ってcontrollerの数分`nil`の入ったlistを返す)
- ただし`:wait-interpolation`に対して`timeout`が実際にresultが返ってくるまでの時間より短く与えられた時にはresultが帰ってきていないcontrollerに対して`t`が返る．

のような挙動にすることになったという理解で良いでしょうか．

また，
`:wait-interpolation-smooth`などの関連メソッドも`:wait-interpolation`に合わせるので問題ないでしょうか．