jsk-ros-pkg/jsk_pr2eus #194: End Effectorをある物体に向けるIKを解く方法

aginika, 2015-12-07:
カメラをarmで把持しているときにそれを被写体に向ける腕のIKを解く方法はあるのでしょうか?
headのlook-atのようなの腕番があるかどうか知りたいです。
end effectorの位置の束縛は必要なしです。

よろしくお願いします.

[image]

snozawa, 2015-12-07:
次のようなものでそこそこ解けるのではと思うのですがいかがでしょうか。
```
(progn                                                                                                                                                                                            
(load "irteus/demo/demo.l")                                                                                                                                                                                   
(setq *robot* (instance sample-robot :init))                                                                                                                                                                  
(send *robot* :reset-pose)                                                                                                                                                                                    
(send *robot* :fix-leg-to-coords (make-coords))                                                                                                                                                               
(objects (list *robot*))                                                                                                                                                                                      
(let ((mt (make-cascoords :coords (send (send *robot* :rarm :end-coords :copy-worldcoords) :rotate pi/2 :y))))                                                                                                
(send (send *robot* :rarm :end-coords :parent) :assoc mt)                                                                                                                                                     
(prog1                                                                                                                                                                                                        
(send *robot* :inverse-kinematics-loop-for-look-at :rarm #f(500 0 0) :debug-view :no-message :move-target mt)                                                                                                 
(send *robot* :inverse-kinematics-loop-for-look-at :rarm #f(500 0 1900) :debug-view :no-message :move-target mt)                                                                                              
(send (send *robot* :rarm :end-coords :parent) :dissoc mt))))
```

おおもとは、look-atしてるメソッドと同じものをよんでいます。
https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L182

特殊な点は
- targetな姿勢を、まいループ視線方向を考慮して計算している(https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L198)
- IKの拘束条件として、:rotation-axis :z、:translation-axis nilにしている（視線方向まわりの回転は許容して、位置誤差も許容する。:translation-axisは:zでもよい）

です。

k-okada, 2015-12-07:
今こうなっているのか．だとすると
send *robot* :rarm :look-at でいけないかな．

snozawa, 2015-12-07:
はい、それでいけます。
むしろその方がわかりやすかったですね。
move-targetの指定は、視線方向によっては必要になります（move-targetのzをむけるため）。

aginika, 2015-12-07:
@snozawa ありがとうございます。

```
(load "pr2eus/pr2-interface.l")
(pr2-init)
(setq a (make-cube 200 200 200))
(send a :translate #f(1500 1500 1500))
(objects (list *pr2* a))
(send *pr2* :ram :look-at (send a :worldpos))
```
の場合、move-targetの指定が無いため,
おっしゃるために以下のようになってしまい。
[image]

move-targetを指定するとできました。
```
(load "package://pr2eus/pr2-interface.l")
(pr2-init)
(setq a (make-cube 200 200 200))
(send a :translate #f(1500 1500 1500))
(objects (list *pr2* a))
(send *pr2* :ram :look-at (send a :worldpos))
(setq mt (make-cascoords :coords (send (send *pr2* :rarm :end-coords :copy-worldcoords) :rotate pi/2 :y)))
(send (send *pr2* :rarm :end-coords :parent) :assoc mt)
(send *pr2* :rarm :look-at (send a :worldpos) :move-target mt)
```
[image]

ありがとうございました。

closed 2015-12-07 by aginika

k-okada, 2015-12-07:
please summarize it in English and add to https://github.com/euslisp/jskeus/issues/318