jsk-ros-pkg/jsk_pr2eus #202: Using :add-controller

snozawa, 2015-12-25:
In the pr2-interface.l, why don't you use `:add-controller` method?
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/pr2-interface.l#L44-L55

This is just a question.

k-okada, 2015-12-27:
@YoheiKakiuchi changed this
https://github.com/jsk-ros-pkg/jsk_pr2eus/commit/de98fede1e12361772b62cc635b17b79042bbe0b

YoheiKakiuchi, 2015-12-30:
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/pr2-interface.l#L44-L55
these code was written before adding keyword ```:create-actions``` to ```:add-controller``` method
https://github.com/jsk-ros-pkg/jsk_pr2eus/commit/1fc74acd5eb1747c11edb151e894bb9f7d9772c1

we can use ```:add-controller``` with ```:create-actions nil``` for PR2 .