jsk-ros-pkg/jsk_pr2eus #203: Why min-time is necessary?

snozawa, 2015-12-26:
@furushchev , @garaemon , @aginika 
I have several questions for min-time in `:angle-vector-duration`:
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L305
- Why min-time is necessary? In `:angle-vector-duration`, `:max-joint-velocity` is used. Isn't it enough?
- Why min-time is set to `1.0` by default? Sometimes old codes include `:angle-vector` smaller than `1.0` [sec] ( https://github.com/jsk-ros-pkg/jsk-ros-pkg-unreleased/blob/master/hrp2_controllers/hrp2_cmd_vel/src/hrp2-cmd-vel-common.l#L184). Constant time `1.0` is too big for small joint difference.

snozawa, 2015-12-26:
In addition, what is the best way to use `:angle-vector` less than `1.0`[sec]?

k-okada, 2015-12-27:
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/121#issuecomment-82948871
said
```

LGTM because

   - staro -> all the programs are old and it does not have scale option
   - hrp2 -> min-time=1.0 and scale=1 is enough

```

:anlge-vector is designed for beginner users to change robot posture and
for expert user who use motion planning or carefully prepared motion will
expect to use angle-vector-sequence.

snozawa, 2015-12-28:
>  :anlge-vector is designed for beginner users to change robot posture and for expert user who use motion planning or carefully prepared motion will expect to use angle-vector-sequence.

I see.
Default value for `min-time` is different between :angle-vector and :angle-vector-sequence, 
but I understand the reason:
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L379

> expert user who use motion planning or carefully prepared motion will expect to use angle-vector-sequence.

In my case, I tested head look-at control by joystick controller, which worked on old OpenHRP2+rats system.
In the code, `:angle-vector` is send with time `200`[msec] and with the :wait-interpolation-smooth.

closed 2015-12-28 by snozawa

k-okada, 2015-12-28:
i see, are you using linear interpolator for this example? did it work
smoothly? what is the parameter for :wait-interpolation-smooth? what was
the minimum rate time for this example?

snozawa, 2015-12-28:
Here is my code:
https://github.com/jsk-ros-pkg/jsk-ros-pkg-unreleased/blob/master/hrp2_controllers/hrp2_cmd_vel/src/hrp2-cmd-vel-common.l#L169-L186 (private repo)
Simplified code is like this:
```
(do-until-key
      (send *ri* :wait-interpolation-smooth 50 :head-controller)
      (send *ri* :angle-vector (send *robot* :angle-vector) 500 :head-controller 0 :min-time 0.0)
  )
```

I send :angle-vector 500[msec] and use :wait-interpolation-smooth with 50[msec].
It works on real HRP2 robot by setting min-time as 0 and I did not tune these time parameter so seriously.
I use hoff-arbib interpolation mode of SequencePlayer in hrpsys-base.

k-okada, 2015-12-29:
what happens if you did not set `:min-time 0` ?

snozawa, 2015-12-29:
Oh, it worked, but it is unexpected behaviour.

On real HRP2, I tried two commands:
```
(bench
(send *ri* :angle-vector (send *robot* :angle-vector) 1000 :head-controller)
(send *ri* :wait-interpolation :head-controller))
;; time -> 1.46333[s]
```
This output is expected.
1.46[s] is more than 1000[msec], so :wait-interpolation successfully wait for finishing :head-controller interpolation.

However, 
```
(bench
(send *ri* :angle-vector (send *robot* :angle-vector) 1000 :head-controller)                                      
(send *ri* :wait-interpolation-smooth 10 :head-controller))
;; time -> 0.015762[s]
```
In this case, expected output is like `;; time -> 0.990[s]` or something.
Returning of `:wait-interpolation-smooth` seem to be too early.

The euslisp test code (https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/test/default-ri-test.l#L35) tests `:wait-interpolation-smooth`, but it is `simulation mode`.
Which part checks the  behaviours of :wait-interpolation and :wait-interpolation-smooth of non-`simulation mode`?