jsk-ros-pkg/jsk_pr2eus #206: pr2 arm-object collision check

iKrishneel, 2016-02-03:
I want the pr2 arm not to collide with the table during manipulation and I tried to use `pqp-collision-check` method. 

The problem is if the links of the arm is inside the cube (representing the table), collision check does not report collision as shown in the attached picture. However if the arm is on the edge of the cube,  collision is reported. 

[Code for this is here: ](https://gist.github.com/iKrishneel/6b5df868b11794cc5aa6)

How to work about this issue?

[image]

k-okada, 2016-02-04:
nice catch, this is very complex problem which caused from conversion between urdf model and eus model. urdf model consists of several links  with rotational joint and fixed joint whereas eus model originally assume a link is consists of one link, not a multiple link connected by fixed joint as urdf do.

see https://gist.github.com/k-okada/2d0e4eacd2f0f5278fa7 for templary workaround (and also i modifed for better visualization, please compare this with yours carefully)

For better solution, we'll continue discussion at -> 
https://github.com/jsk-ros-pkg/jsk_model_tools/issues/171

chiwunau, 2016-05-24:
I have a similar problem, where I am trying to solve IK while avoiding the robot's arm and base collide with the door model. The screenshot below shows the result when I put every pair of door links and links from ```(send *pr2* :link-list (send *pr2* :rarm :end-coords :parent))``` to ```:collision-avoidance-link-pair``` of IK. ([code](https://github.com/chiwunau/pr2_open_b2_door_demo/blob/contact-search/euslisp/contact_state/search-motion.l#L29))
It seem that when searching in [null-space](https://github.com/euslisp/jskeus/blob/master/irteus/demo/null-space-ik.l), It only consider collision of bodies in the link-list. In this case many parts of arm and base are missing.
[image]

k-okada, 2016-06-06:
@chiwunau @iKrishneel 
can we try "workaround" for this issue?
1) create list contains "body object" for each links, manually
2) create convex hull of these objects
3) replace collision object for each links
4) write this procedure within pr2-utils.l
@furushchev can you help them?

chiwunau, 2016-06-13:
Thank you, I will get @furushchev help and try writing the code.

chiwunau, 2016-06-14:
This change (https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/232) can fix the problem, I tested for IK and ```pqp-collision-check```.
here is the [test code](https://gist.github.com/chiwunau/e78698d9906d8c027c003337748d9153)