jsk-ros-pkg/jsk_pr2eus #207: how to run pr2-tabletop-demo

k-okada, 2016-02-09:
@aginika , please add documentation in the code,(which launch file is required, how to use this code)
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus_moveit/tutorials/pr2-tabletop-demo.l
@iKrishneel

aginika, 2016-02-09:
Ok, I will

furushchev, 2016-04-06:
Created PR.
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/216
Tested on PR1012.
Now creating test codes

k-okada, 2016-04-06:
- [ ] test cdoe
- [ ] add more documents to https://github.com/jsk-ros-pkg/jsk_pr2eus/tree/master/pr2eus_tutorials

furushchev, 2016-04-09:
add test code and document
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/217

k-okada, 2016-05-28:
@iKrishneel  https://github.com/jsk-ros-pkg/jsk_roseus/pull/436 may include how to run pr2 tabletop demo

iKrishneel, 2016-07-01:
@k-okada sensei,

Is there any tutorial on dual arm manipulation of PR2. Eg. Moving large box which cannot be grasped via use of both arms pushing on each side of the box. How can  I solve such manipulation?