jsk-ros-pkg/jsk_pr2eus #227: controller をadd-controllerで上書きした時の話

pazeshun, 2016-05-23:
```
       ;;                                                                                                                                             
       (let ((cacts (gethash ctype controller-table)))
         (print (list 'cacts cacts ctype controller-table))
         (unless cacts
           (warn ";; controller-type: ~A not found" ctype)
           (return-from :angle-vector-sequence))
         (mapcar
          #'(lambda (action param)
              (print (list 'action action param))
              (send self :send-ros-controller
                    action (cdr (assoc :joint-names param)) ;; action server and joint-names                                                          
                    start-time  ;; start time                                                                                                         
                    traj-points))
          cacts (send self ctype)))
```

:angle-vectorはだいたい上のようになっているのですが、このとき
(gethash ctype controller-table)と(send self ctype)で帰ってくるコントローラの順番が一緒になることが想定されている気がします。一方で、add-controllerで
```
      (t ;; not creating actions, just search                                                                                                         
       (mapcar
        #'(lambda (param)
            (let* ((controller-action (cdr (assoc :controller-action param)))
                   (action-type (cdr (assoc :action-type param)))
                   (name (if namespace (format nil "~A/~A" namespace controller-action)
                           controller-action))
                   action)
              (setq action (find-if #'(lambda (ac) (string= name (send ac :name)))
                                    controller-actions))
              (if action (push action tmp-actions))
              ))
```
で一回できているcontrollerについて変更した場合、pushしているので順番が逆順になっているきがします。
よく見ると、
https://github.com/jsk-ros-pkg/jsk_pr2eus/blame/master/pr2eus/robot-interface.l#L202
ではnreverseしているけど、その後の
https://github.com/jsk-ros-pkg/jsk_pr2eus/blame/master/pr2eus/robot-interface.l#L233
当たりではreverseしていないのが間違いでしょうか？　@YoheiKakiuchi 

by  @k-okada

k-okada, 2016-05-23:
ki#228 で対応しました．確認してみて下さい @YoheiKakiuchi

YoheiKakiuchi, 2016-05-23:
すいません。 そのとおりです。
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/228 が正しいです。

k-okada, 2016-05-24:
確認ありがとうございます．

closed 2016-05-24 by k-okada

pazeshun, 2016-05-24:
これは、いつリリースされるのでしょうか

k-okada, 2016-05-24:
機能リリースしたばっかりなので，すこし時間開けたいところですが，どうでしょうか．
jsk_apc 側は，昨日の変更コードをいれておけば問題はないと思います．

pazeshun, 2016-05-24:
確認したかっただけなので、いつでも大丈夫です。

PR #228 (2016-05-24): robot-interface.l : need a consistency of controller order in the the entry of controller-table fix #227
Close #227
commit 0b634363 (2016-05-24): Merge pull request #228 from k-okada/fix_227

robot-interface.l : need a consistency of controller order in the the entry of controller-table fix #227
commit 999eea5a (2016-05-24): robot-interface.l : need a consistency of controller order in the the entry of controller-table fix #227