jsk-ros-pkg/jsk_pr2eus #231: wait-interpolation-smooth method never returns

pazeshun, 2016-06-01:
I want to use `:wait-interpolation-smooth` of `robot-interface.l` in pr2eus package.
When I tried to test this method, I found that this method never returned even after the motion of the robot ended.

I used real Baxter and start-jsk/jsk_apc package.
I didn't test enough, so I'll put more detailed information after debugging.

k-okada, 2016-06-01:
please provide following information,
```
rostopic echo <your controller>/goal
rostopic echo <your controller>/status
rostopic echo <your controller>/feedback
rostopic echo <your controller>/result
```

pazeshun, 2016-06-02:
```
$ rostopic echo /robot/limb/right/follow_joint_trajectory/goal
header: 
  seq: 1
  stamp: 
    secs: 1464869531
    nsecs: 635470495
  frame_id: ''
goal_id: 
  stamp: 
    secs: 0
    nsecs:         0
  id: 1464869531636058622_/robot_main_1464869123488549266_15548_/robot/limb/right/follow_joint_trajectory_1
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 1464869531
        nsecs: 635470495
      frame_id: ''
    joint_names: ['right_s0', 'right_s1', 'right_e0', 'right_e1', 'right_w0', 'right_w1', 'right_w2']
    points: 
      - 
        positions: [-1.7011846392236354, -0.041800984651114564, 1.650179137163354, 2.350434903660762, -1.595339146736439, 0.1518640652757549, 0.0]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 0
          nsecs:  50000000
      - 
        positions: [-1.7011846392236354, -0.041800984651114564, 1.650179137163354, 2.350434903660762, -1.595339146736439, 0.1518640652757549, 0.0]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 15
          nsecs:  50000000
      - 
        positions: [-1.7011846392236354, -0.041800984651114564, 1.650179137163354, 2.350434903660762, -1.595339146736439, 0.1518640652757549, 0.0]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 30
          nsecs:  50000000
      - 
        positions: [-1.7011846392236354, -0.041800984651114564, 1.650179137163354, 2.350434903660762, -1.595339146736439, 0.1518640652757549, 0.0]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 30% rostopic echo /robot/limb/right/follow_joint_trajectory/goal
header: 
  seq: 1
  stamp: 
    secs: 1464869531
    nsecs: 635470495
  frame_id: ''
goal_id: 
  stamp: 
    secs: 0
    nsecs:         0
  id: 1464869531636058622_/robot_main_1464869123488549266_15548_/robot/limb/right/follow_joint_trajectory_1
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 1464869531
        nsecs: 635470495
      frame_id: ''
    joint_names: ['right_s0', 'right_s1', 'right_e0', 'right_e1', 'right_w0', 'right_w1', 'right_w2']
    points: 
      - 
        positions: [-1.7011846392236354, -0.041800984651114564, 1.650179137163354, 2.350434903660762, -1.595339146736439, 0.1518640652757549, 0.0]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 0
          nsecs:  50000000
      - 
        positions: [-1.7011846392236354, -0.041800984651114564, 1.650179137163354, 2.350434903660762, -1.595339146736439, 0.1518640652757549, 0.0]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 15
          nsecs:  50000000
      - 
        positions: [-1.7011846392236354, -0.041800984651114564, 1.650179137163354, 2.350434903660762, -1.595339146736439, 0.1518640652757549, 0.0]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 30
          nsecs:  50000000
      - 
        positions: [-1.7011846392236354, -0.041800984651114564, 1.650179137163354, 2.350434903660762, -1.595339146736439, 0.1518640652757549, 0.0]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 30
          nsecs: 100000000
  path_tolerance: []
  goal_tolerance: []
  goal_time_tolerance: 
    secs: 0
    nsecs:         0
---

          nsecs: 100000000
  path_tolerance: []
  goal_tolerance: []
  goal_time_tolerance: 
    secs: 0
    nsecs:         0
---
```
```
$ rostopic echo /robot/limb/right/follow_joint_trajectory/result
header: 
  seq: 2
  stamp: 
    secs: 1464869561
    nsecs: 850416898
  frame_id: ''
status: 
  goal_id: 
    stamp: 
      secs: 1464869531
      nsecs: 637715101
    id: 1464869531636058622_/robot_main_1464869123488549266_15548_/robot/limb/right/follow_joint_trajectory_1
  status: 3
  text: ''
result: 
  error_code: 0
  error_string: ''
---
```
`status` and `feedback` are too long to see.

k-okada, 2016-06-02:
for long file you can save to the file
`rostopic echo /hogehoge | tee hogehoge.txt`
and upload to gist.github.com
￼

pazeshun, 2016-06-02:
Now I realize that gripper joint trajectory action has problem.
So it is the only problem for Baxter for jsk_apc.

pazeshun, 2016-06-02:
This package has no problem, so close this issue.
Move to start-jsk/jsk_apc#1624

closed 2016-06-02 by pazeshun