jsk-ros-pkg/jsk_pr2eus #233: :wait-interpolation-smoothの待機時間

k-kimura, 2016-06-17:
```
(send *ri* :wait-interpolation-smooth time-to-finish &optional (ctype))
```
の使い方ですが，
http://euslisp-docs.readthedocs.io/en/latest/pr2eus/robot-interface/#wait-interpolation-smooth
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L514-521
などを参考にすると，
直前に送った指令(angle-vectorなど)が終了する```time-to-finish```[msec]前まで，
:wait-interpolationで待機する，という理解で合ってますでしょうか．

この理解を前提として，実機(シミュレータでない)の\*ri\*に以下のコマンドを送って時間を測ってみると，
約4[s]くらいの値が返ってくる予想となりますが，ほとんど待機なしの早い値が返ってくる結果となりました．
(ちなみに,今回の例では\*hrp2jsk\*実機を用いてreset-pose -> reset-manip-poseのangle-vector遷移を参考としました．)
```
irteusgl$ (bench
(send *ri* :angle-vector (send *robot* :angle-vector) 5000) ;; 5[s]で遷移
(send *ri* :wait-interpolation-smooth 1000) ;; 遷移終了の1[s]前で:wait-interpolationから抜ける?
)
;; time -> 0.009716[s]
nil
```

k-kimura, 2016-06-17:
また，参考までの情報ですが，
- :wait-interpolation-smooth
- :wait-interpolation
- unix:usleep

を使って，各実行時間を簡単に比較しました．
(いずれも\*hrp2jsk\*実機でのreset-pose -> reset-manip-poseの変化です．)

```
irteusgl$ (bench (send *ri* :angle-vector (send *robot* :angle-vector) 5000) (send *ri* :wait-interpolation-smooth 50))
;; time -> 0.018292[s]
nil

irteusgl$ (bench (send *ri* :angle-vector (send *robot* :angle-vector) 5000) (send *ri* :wait-interpolation-smooth 1000))
;; time -> 0.020518[s]
nil

irteusgl$ (bench (send *ri* :angle-vector (send *robot* :angle-vector) 5000) (send *ri* :wait-interpolation-smooth 5000))
;; time -> 0.023807[s]
nil

irteusgl$ (bench (send *ri* :angle-vector (send *robot* :angle-vector) 5000) (send *ri* :wait-interpolation))
;; time -> 5.22598[s]

irteusgl$ (bench (send *ri* :angle-vector (send *robot* :angle-vector) 5000) (unix:usleep (* 5000 1000))) ;; unix:usleepを使って5[s]だけ待機
;; time -> 5.0199[s]

irteusgl$ (bench (send *ri* :angle-vector (send *robot* :angle-vector) 5000) (unix:usleep (round (* 0.8 (* 5000 1000))))) ;; unix:usleepを使って0.8*5[s]だけ待機
;; time -> 4.0195[s]
```

k-okada, 2016-06-18:
それはrosbridgeがなにか間違えている可能性が高いです．
:angle-vectorをおくるとjointtrajectoryactoinを送る，という部分はわかっているかな．
であれば，そのactionのfeedbackを見て下さい．

https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L514-L533
から
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L40-L52
あたりで経過時間をしらべています．

k-okada, 2016-06-20:
feedback でていないですね．

k-kimura, 2016-06-20:
デバッグ中ですが，
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L533
まできたときに
```
(0)
```
のリストが返ってきています．

ここでのcontroller-actionsは
```
(#<controller-action-client #X876a2f0 fullbody_controller/joint_trajectory_action>)
```
のリストで
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L40-L42
において```time-to-finish```をはき出していますが，
前の
```
(setq time-to-finish 0)
```
がそのまま返ってきている様子です．

ros_bridgeを立ち上げた後に
```
rostopic echo /fullbody_controller/joint_trajectory_action/feedback
```
を確認するのが正しいか怪しいですが，
手元のhrpsys simulator + hrp2実機(hrp20xxv)共に確かにこのトピックは出ていないですね．

k-kimura, 2016-06-20:
@pazeshun 
:wait-interpolation-smoothで待つことができるというBaxter系の環境では
```
/fullbody_controller/joint_trajectory_action/feedback
```
のトピックははき出されていますでしょうか？

k-okada, 2016-06-20:
なるほど．これは案外難しかったですね．
https://github.com/start-jsk/rtmros_common/pull/957
して
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/235
したらいけるとおもいます．ところで，いまつかっているのはfullbody？fullbodyだけpr2_controllers_msgsになって，larm_controllerとかつかうとcontrollers_msgs
になるのかな？
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L773
を直してもいいのかな？

@pazeshun の問題は
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/236
で解決するかな．

pazeshun, 2016-06-20:
@k-kimura 
Baxter環境では、`/robot/limb/right/follow_joint_trajectory/feedback`とかなのですが、これはちゃんと出ていることを確認しています。一時期グリッパーの方のjtaのfeedbackがまちがっていたのですが、その時は永遠に終わらなかったです。( #231 )

k-kimura, 2016-06-20:
start-jsk/rtmros_common#957
#235
の変更を加えたところ，angle-vectorを送っている間だけ
```
rostopic echo /fullbody_controller/joint_trajectory_action/feedback
```
のfeedbackが返ってきました．
(hrpsys simulatorで確認)

一方で:wait-interpolation-smoothの待ち時間はまだ早いままなので，
time-to-finishあたりを確認してみます．
```
irteusgl$ (bench (send *ri* :angle-vector (send *robot* :angle-vector) 5000) (send *ri* :wait-interpolation-smooth 2000))
;; time -> 0.149312[s]
nil
```

> ところで，いまつかっているのはfullbody？

はい，現状は
```
(send *ri* :wait-interpolation-smooth 2000)
```
としてctype(controller-type)を特に指定せずに
fullbodyでチェックしています．
```
irteusgl$ (send *ri* :get-val 'controller-actions)
(#<controller-action-client #X8250c40 fullbody_controller/joint_trajectory_action>)
```

また，@pazeshun への#236 の変更も加えて試すと，
#231 のissueと同じく永久に終わらない症状が出てしまいました．

k-okada, 2016-06-20:
#235 と #236 と両方含まれているコードをつかっている？

k-kimura, 2016-06-20:
はい，#235 と #236 の両方の変更を加えたrobot-interface.lをloadして
```
irteusgl$ (bench (send *ri* :angle-vector (send *robot* :angle-vector) 5000) (send *ri* :wait-interpolation-smooth 1000))
```
を試すと，returnせずに抜け出せない状況です．
(```/fullbody_controller/joint_trajecty_action/feedback```は5秒間だけ流れてきています．)

k-okada, 2016-06-20:
ループしているのは
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/236/files#diff-15f6920b924513f6027d36b361f85bbdR527
だろうか．

 (while (some #'(lambda (x) (print (list tm (send x
:last-feedback-msg-stamp)) (<= (send (ros::time- (send x
:last-feedback-msg-stamp) tm) :to-sec) 0)) cacts)
にしたらどうなるかな＞

k-kimura, 2016-06-20:
ループ箇所を変更した後の結果です．
```
irteusgl$ (bench (send *ri* :angle-vector (send *robot* :angle-vector) 5000) (send *ri* :wait-interpolation-smooth 1000))
(#<ros::time #X934aeb8 5732.027> #<ros::time #X8323340 5706.815>)
(#<ros::time #X934aeb8 5732.027> #<ros::time #X8323340 5706.815>)
(#<ros::time #X934aeb8 5732.027> #<ros::time #X8323340 5706.815>)
(#<ros::time #X934aeb8 5732.027> #<ros::time #X8323340 5706.815>)
(#<ros::time #X934aeb8 5732.027> #<ros::time #X8323340 5706.815>)
(#<ros::time #X934aeb8 5732.027> #<ros::time #X8323340 5706.815>)
(#<ros::time #X934aeb8 5732.027> #<ros::time #X8323340 5706.815>)
(#<ros::time #X934aeb8 5732.027> #<ros::time #X8323340 5706.815>)
(#<ros::time #X934aeb8 5732.027> #<ros::time #X8323340 5706.815>)
(以下ループ)
```

k-okada, 2016-06-20:
あ，
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/236/files#diff-15f6920b924513f6027d36b361f85bbdR47
の後ろに
(setq last-feedback-msg-stamp (ros::time-now))
がないとダメじゃない．

k-okada, 2016-06-20:
ちがうや，
  (:action-feedback-cb (msg)
   (let ((finish-time) (current-time))
     (setq last-feedback-msg-stamp (send msg :header :stamp))
     (ros::ros-debug "[~A] feedback-cb ~A" ros::name-space msg)
が正解

k-okada, 2016-06-20:
さらに変更しました．バグをいれていないか確認してもらえると嬉しいです @k-kimura

k-kimura, 2016-06-21:
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/236/commits/a8923e549954f62147ceb1ca51ea38009d841e0c の変更を加えても大丈夫そうでした．

pazeshun, 2018-07-09:
@k-kimura 現在使っているHRP2ではFullbodyでもFollowJointTrajectoryActionを使っていますでしょうか。もしそうなら、`:wait-interpolation-smooth`が動かなくなっているのではないかと思うのですが、どうでしょうか。

k-kimura, 2018-07-09:
最近更新した最新版のHRP2環境では，
angle-vector送ったときに
```/fullbody_controller/joint_trajectory_action/feedback```
ではなく
```/fullbody_controller/follow_joint_trajectory_action/feedback```
のactionlibが応答していて，確かに```follow_joint_trajectory_action```の場合は
:wait-interpolation-smoothから永遠に抜けられないようです．

以前に自分が:wait-interpolation-smoothが使えるのを確認したのは
まだ```joint_trajectory_action```をdefault-controllerとして採用していた頃で
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/237
の議論を経て，
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/250
の変更が加わってからは```follow_joint_trajectory_action```を使うようになったので
:wait-interpolation-smoothが動かなくなってしまったんだとおもいます．

pazeshun, 2018-07-09:
@k-kimura https://github.com/start-jsk/rtmros_common/pull/1049 で多分治ったと思うので、試していただけると嬉しいです。

pazeshun, 2018-07-13:
いろいろ考えた末、 #356 を作りました。
これで、https://github.com/start-jsk/rtmros_common/pull/1049 なしでも`:wait-interpolation-smooth`が動くと思います。
ただ、https://github.com/start-jsk/rtmros_common/pull/1052 なしだと、`:angle-vector`に`start-time`を渡した場合、動作では無視されるが`:wait-interpolation-smooth`では無視されず、ずっと待機する挙動になります。