jsk-ros-pkg/jsk_pr2eus #234: [pr2eus_moveit] planning with base

furushchev, 2016-06-19:
I'm totally a moveit newbie.
I found the config file of pr2:

```xml
    <group name="whole_body">
        <group name="base" />
        <group name="arms" />
        <group name="torso"/>
    <group/>
```

refer: https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus_moveit/euslisp/pr2eus-moveit.l#L81-L82

Does it mean we can plan a collision aware motions including not only arm / torso but also base?
If so, how do I use `pr2eus_moveit` to plan with base?
@YoheiKakiuchi @k-okada

YoheiKakiuchi, 2016-06-20:
> Does it mean we can plan a collision aware motions including not only arm / torso but also base?

yes

> If so, how do I use pr2eus_moveit to plan with base?

not implemented yet

k-okada, 2016-06-20:
On Sun, Jun 19, 2016 at 6:46 PM, Furushchev <notifications@github.com>
wrote:

> If so, how do I use pr2eus_moveit to plan with base?
>

I think so , but it may not so usefull, i think

--
◉ Kei Okada

YoheiKakiuchi, 2016-06-20:
> it may not so usefull, i think

+1

furushchev, 2016-06-20:
@YoheiKakiuchi @k-okada Could you explain why you think so? I don't have such an impression at all now.

k-okada, 2016-06-20:
- from theroy navigaiton planning and motion planing are same
- so if we create C-space of base, you can create navigation planning ->
出来ている
- but if you try to run this on real robot you have to care about many
thinngs, speed/accel limitation, recovery mode....
- and I haven't seen anyone using this feature on real robots -> 出来ていない