jsk-ros-pkg/jsk_pr2eus #248: [pr2eus] Freeze in :wait-until-update-all-joints in simulation mode

pazeshun, 2016-08-31:
This happened in start-jsk/jsk_apc
I don't know about other repos.

k-okada, 2016-09-01:
does this happen when you create baxter-interface, just loaded from baxtereus/baxter-interface.l ? @pazeshun

mmurooka, 2016-09-01:
It works in my environment.

Please download https://gist.github.com/mmurooka/6906e6997b0be948b58eef35950a2443 , and execute
```
roscore
roseus test-baxter-interface-wait-until-update-all-joints.l 
```

My output is following, and it is working correctly.
```
":wait-until-update is nil"
;; time -> 2.500000e-05[s]
":wait-until-update is t"
;; time -> 3.300000e-05[s]
":wait-until-update is nil"
;; time -> 4.000000e-06[s]
":wait-until-update is t"
;; time -> 2.300000e-05[s]
":wait-until-update is nil"
;; time -> 3.000000e-06[s]
":wait-until-update is t"
;; time -> 2.100000e-05[s]
```

pazeshun, 2016-09-01:
> does this happen when you create baxter-interface, just loaded from baxtereus/baxter-interface.l ?

Yes. 
@mmurooka's `test-baxter-interface-wait-until-update-all-joints.l` works in my environment like following:
```
":wait-until-update is nil"
;; time -> 1.400000e-05[s]
":wait-until-update is t"
;; time -> 3.300000e-05[s]
":wait-until-update is nil"
;; time -> 4.000000e-06[s]
":wait-until-update is t"
;; time -> 2.700000e-05[s]
":wait-until-update is nil"
;; time -> 4.000000e-06[s]
":wait-until-update is t"
;; time -> 2.500000e-05[s]
```
But when I typed `send *ri* :update-robot-state :wait-until-update t` to roseus command line, it never returned:
```
1.irteusgl$ send *ri* :update-robot-state :wait-until-update t

```

k-okada, 2016-09-16:
@pazeshun can you create test code for this?