jsk-ros-pkg/jsk_pr2eus #258: /base_footprint基準で回転したcoordsを:move-toに指示するとaction結果が返ってこない

h-kamada, 2016-11-13:
`/base_footprint`をframe-idとして、回転を加えたgoalを指定すると
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L1253
において結果が返ってこなくなる。回転を与えなければ返ってくる。
send-goal の値を見ると、回転した場合z成分の値が微妙に元の値と変わっていることが分かる。
zの値が地図がセットされている平面と異なるために同じ階層として認識出来ていない、ということでしょうか？
```
2.irteusgl$ send *ri* :move-to (send (make-coords) :rotate 0 :z :world) :frame-id "/base_footprint"
[ INFO] [1479048095.377515181]: move-to : send-goal to #f(5104.13 6236.03 -0.226538) at /map (0)
[ INFO] [1479048095.377966180]: before wait-for-result
[ INFO] [1479048095.678178139]: after wait-for-result
[ INFO] [1479048095.678259594]: move-to : succeeded
[ WARN] [1479048095.729482010]: :move-to wait-for transform /map to /base_footprint -> t
[ WARN] [1479048095.729700396]: :move-to current-coords  #<cascaded-coords #X7600250 /map  5104.13 6236.025 -0.227 / 0.128 -0.023 -0.001>
[ WARN] [1479048095.729755297]:          mapgoal-coords  #<coordinates #X76fdd48  5104.13 6236.025 -0.227 / 0.128 -0.022 -0.001>
[ WARN] [1479048095.729801196]:           error-coords   #<coordinates #X7600bb0  -5.551e-06 6.059e-06 1.053e-07 / 2.841e-06 0.0 -0.0>
[ WARN] [1479048095.729844449]:          target-coords   #<coordinates #X7ca42f0  0.0 0.0 0.0 / 0.0 0.0 0.0>
[ WARN] [1479048095.729900989]: :move-to -> :go-velocity x:-0.067552 y:0.073734 d:34.57 msec:8.217419e-05
[ WARN] [1479048096.130117675]: :go-velocity is disabled. (move-base-trajectory-action is not found)
[ WARN] [1479048096.221065174]: :move-to wait-for transform /map to /base_footprint -> t
[ WARN] [1479048096.221265958]: :move-to current-coords  #<cascaded-coords #X7724c48 /map  5104.13 6236.025 -0.227 / 0.128 -0.022 -0.002>
[ WARN] [1479048096.221341023]:          mapgoal-coords  #<coordinates #X7724c90  5104.13 6236.025 -0.227 / 0.128 -0.022 -0.001>
[ WARN] [1479048096.221390727]:           error-coords   #<coordinates #X77268c8  -1.425e-05 1.556e-05 2.731e-07 / 3.295e-07 3.897e-05 4.483e-05>
[ WARN] [1479048096.221435503]:          target-coords   #<coordinates #X7ca42f0  0.0 0.0 0.0 / 0.0 0.0 0.0>
[ WARN] [1479048096.221486813]: :move-to -> :go-velocity x:-0.067553 y:0.073734 d:1.56162 msec:0.000211
[ WARN] [1479048096.621629144]: :go-velocity is disabled. (move-base-trajectory-action is not found)
[ WARN] [1479048096.713004051]: :move-to wait-for transform /map to /base_footprint -> t
[ WARN] [1479048096.713476488]: :move-to current-coords  #<cascaded-coords #X77bdac8 /map  5104.13 6236.025 -0.227 / 0.128 -0.022 -0.001>
[ WARN] [1479048096.713596376]:          mapgoal-coords  #<coordinates #X77bf880  5104.13 6236.025 -0.227 / 0.128 -0.022 -0.001>
[ WARN] [1479048096.713652058]:           error-coords   #<coordinates #X77bdc78  -9.225e-06 1.007e-05 1.697e-07 / -3.392e-06 -0.0 -0.0>
[ WARN] [1479048096.713694452]:          target-coords   #<coordinates #X7ca42f0  0.0 0.0 0.0 / 0.0 0.0 0.0>
t
3.irteusgl$ send *ri* :move-to (send (make-coords) :rotate 1 :z :world) :frame-id "/base_footprint"
[ INFO] [1479048100.671565472]: move-to : send-goal to #f(5104.13 6236.03 -0.226537) at /map (0)
[ INFO] [1479048100.671885242]: before wait-for-result
  C-c C-cinterrupt4.B1-irteusgl$   C-c C-cinterrupt4.B2-irteusgl$ 
```

h-kamada, 2016-11-14:
今日もう一度試したら問題なく動きました。

closed 2016-11-14 by h-kamada

k-okada, 2016-11-16:
> において結果が返ってこなくなる。回転を与えなければ返ってくる。

はactionlibから帰ってこなかったのかな・