jsk-ros-pkg/jsk_pr2eus #262: Testing codes including speak functions on non-real-robot environment.

snozawa, 2016-11-24:
Recently, `speak-jp` and `speak-en` functions are using actionlib:
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/241

Previously, codes including speak functions can be tested on non-real-robot environment, e.g., no `robotsound` actionserver.
My questions are:
- After using actionlib, do we need to start up `robotsound` on non-real-robot environment?
- Do we need to use sound_play from source? Is the actionlib version provided from deb?
- How do you test pr2 codes on non-real-robot testing environment?

@furushchev

furushchev, 2016-11-24:
@snozawa Answered inline:

> After using actionlib, do we need to start up robotsound on non-real-robot environment?

No.

> Do we need to use sound_play from source? Is the actionlib version provided from deb?

from source and my pull request. Currently there are bug both from source and from apt, I think.
I mean "I think", because the maintainer does not admit there is a bug and not merge my pull requests.
I will handle these issues when I have time.

> How do you test pr2 codes on non-real-robot testing environment?

Please see https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/246.
If `(setq *speak-timeout* 0)` is declared in test codes, speak functions should be ignored without launching `sound_play` server.

snozawa, 2016-11-24:
Thanks!
I got the all answers.

closed 2016-11-24 by snozawa