jsk-ros-pkg/jsk_pr2eus #271: pr2の脱力モード

furushchev, 2016-12-08:
pr2で片腕のみ脱力させる、といったことをしようとしているのですが、pr2 teleop joyはどのように マネキンと脱力とサーボを切り替えているのでしょうか。（コードはどこにありますか？）
pr2_teleop_general_joystick.launchを見ているのですが、どこが該当しているのか見つけられていません。

by @h-kamada

furushchev, 2016-12-08:
だいぶ前のコードですが、僕はこんな感じで切り替えて使っていた覚えがあります。

https://github.com/jsk-ros-pkg/euslib/blob/master/demo/furushchev/ros_local_repo/local_pr2_demo/euslisp/test-util.l

h-kamada, 2016-12-08:
なるほど、ありがとうございます。
こちらのサイトも読んでいたのですが、
http://wiki.ros.org/pr2_controller_manager
switch controllerのサービスはあくまでも何かのcontrollerから別へ変えるものだと思い込んでました。
が、よく読んでみるとstart引数をnilにも出来るようなのでこれで良いのですね！ありがとうございます。

furushchev, 2016-12-08:
@h-kamada 落ち着いたらでいいので、使えることがわかったコードを綺麗な形でpr2eusにプルリクエストシていただけると助かります。

k-okada, 2016-12-12:
https://github.com/PR2/pr2_apps/tree/hydro-devel/pr2_mannequin_mode/launch

コントローラ変えているみたいですね．

furushchev, 2017-01-24:
@h-kamada ping

h-kamada, 2017-01-24:
こちらの方忘れてました。
修論の際、eusからコントローラを変えようとした際に変なエラーが出て、仕方ないのでコマンドラインからswitchする形でやり繰りしてました。
修論も落ち着いたので、もう一度eusからswitchするコードを書いてみます。

furushchev, 2017-05-25:
This feature is now available from https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/295

closed 2017-05-25 by furushchev