jsk-ros-pkg/jsk_pr2eus #274: speak.l がsourceのsound_playでは動かない

ban-masa, 2016-12-23:
sourceのsound_playではSoundRequest.msgにvolumeが追加されていて、defaultで０になっているようです。
https://github.com/ros-drivers/audio_common/blob/master/sound_play/msg/SoundRequest.msg
そのためspeak.lでmsgを送ってもvolume0で再生されてしまって聞こえません。

speak.lをvolumeを設定するように変更したい気がするのですが、そうするとindigoのdevのsound_play（volumeパラメータなし）で使えなくなると思います。

この場合sourceとdevのどちらで動くようにするべきなのでしょうか。
それとも両方で動くようにslot変数を見て条件分岐させたりするべきでしょうか。

k-okada, 2016-12-24:
source を使う必要はなにがあるでしょうか？
特にいないなら、debでいいと思います．どうしてもソースを使いたいなら、
https://github.com/ros-drivers/audio_common/blob/indigo-devel/sound_play/msg/SoundRequest.msg
のblanchを使うのがいいです．

jade/kineticだとこっちのmsgが使われ得ているのかな．であるとバージョンでspeak.l のコードの中で切り替える必要がでてきますね．

ban-masa, 2016-12-24:
なぜかhrp2016でsourceが使われていたので一時的にhrp2016が喋らなくなっていました。
sourceのmasterを使う必要性は何もないはずなので、indigo-develに切り替えるかdevを使うように変更しようと思います。
ありがとうございました。

closed 2016-12-24 by ban-masa

snozawa, 2016-12-25:
> source を使う必要はなにがあるでしょうか？特にいないなら、debでいいと思います．どうしてもソースを使いたいなら、

sourceを使うのは、speak.lでwait-for-serverが使われるようになって、それに対応できる内容がdebに反映されてなかった（当時）ので、sourceにしました。
なので、indigo-develをつかうのがよさそうですね。
debに反映されればsourceを使う必要がなくなります。
（もしくは、soud_playをsourceするのでなく、最近しった`(setq *speak-timeout* 0)`でも良かったかもしれません）

> jade/kineticだとこっちのmsgが使われ得ているのかな．であるとバージョンでspeak.l のコードの中で切り替える必要がでてきますね．

また、speak.l側はこれでも良かったとも思います。

k-okada, 2016-12-25:
sound play のactionlib対応は0.2.9ぐらいで行われていそうで、リリースされているのは0.2.12なので、大丈夫そうです．
==
https://github.com/ros-drivers/audio_common/blob/indigo-devel/sound_play/CHANGELOG.rst
http://repositories.ros.org/status_page/ros_indigo_default.html?q=audio_common

また、
> sourceを使うのは、speak.lでwait-for-serverが使われるようになって、

は、JSKのソフトは同じソースツリーで、hydro/indigoのサポートをしてきているので、indigoにしないと動かない、というものは作ってきていないはずなので、そういう影響があったら、ソースツリーで最新にするのではなくて、文句を言って良いレベルでしたね．

snozawa, 2016-12-26:
なるほど、それでは確認して動けばdebで良いと思います。

> は、JSKのソフトは同じソースツリーで、hydro/indigoのサポートをしてきているので、indigoにしないと動かない、というものは作ってきていないはずなので、そういう影響があったら、ソースツリーで最新にするのではなくて、文句を言って良いレベルでしたね．

わかりました。

ban-masa, 2016-12-26:
hrp2016でaudio_commonをdebで入れてspeak.lが問題なく動作することを確認しました。