jsk-ros-pkg/jsk_pr2eus #294: A lot of build-failed cron branch

knorth55, 2017-05-12:
There are a lot of cron branch in `origin`
How about using cron function in Travis?

```
$ git fetch origin
Fetching origin
gPOST git-upload-pack (gzip 2147 to 755 bytes)
remote: Counting objects: 232, done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 232 (delta 127), reused 126 (delta 126), pack-reused 96
Receiving objects: 100% (232/232), 72.52 KiB | 92.00 KiB/s, done.
Resolving deltas: 100% (166/166), completed with 28 local objects.
From https://github.com/jsk-ros-pkg/jsk_pr2eus
   e3d928f..53e38ba  master     -> origin/master
 = [up to date]      cron_for_github/01da8ab0-4c29-4219-8cf5-0184ba9e9401 -> origin/cron_for_github/01da8ab0-4c29-4219-8cf5-0184ba9e9401
 * [new branch]      cron_for_github/0432d270-f74e-408e-b539-7e75238f6719 -> origin/cron_for_github/0432d270-f74e-408e-b539-7e75238f6719
 = [up to date]      cron_for_github/0777963e-05da-4b7a-960e-d4f2951a0e32 -> origin/cron_for_github/0777963e-05da-4b7a-960e-d4f2951a0e32
 * [new branch]      cron_for_github/0a016517-b9ec-410e-a69a-c3d036a8df2e -> origin/cron_for_github/0a016517-b9ec-410e-a69a-c3d036a8df2e
 * [new branch]      cron_for_github/0aaed286-dc3c-4ef9-8f40-5d3e8ee9ed87 -> origin/cron_for_github/0aaed286-dc3c-4ef9-8f40-5d3e8ee9ed87
 * [new branch]      cron_for_github/0d10ee2d-272a-4ee1-8aa1-108b38b131a7 -> origin/cron_for_github/0d10ee2d-272a-4ee1-8aa1-108b38b131a7
 * [new branch]      cron_for_github/142c8709-9657-40b5-8dbb-4d5606ce72b2 -> origin/cron_for_github/142c8709-9657-40b5-8dbb-4d5606ce72b2
 * [new branch]      cron_for_github/197fe14d-e190-48a9-a507-e9618b4ce20c -> origin/cron_for_github/197fe14d-e190-48a9-a507-e9618b4ce20c
 * [new branch]      cron_for_github/1a875de3-12d8-413e-af5a-7cdf30df5450 -> origin/cron_for_github/1a875de3-12d8-413e-af5a-7cdf30df5450
 = [up to date]      cron_for_github/2c08aa34-f711-4615-94f6-d192cfbb8b33 -> origin/cron_for_github/2c08aa34-f711-4615-94f6-d192cfbb8b33
 = [up to date]      cron_for_github/2c58d9a8-11df-48ac-b7c2-82acfef972c1 -> origin/cron_for_github/2c58d9a8-11df-48ac-b7c2-82acfef972c1
 = [up to date]      cron_for_github/3d23cf6c-4c93-40ea-a4cc-c91b8143f103 -> origin/cron_for_github/3d23cf6c-4c93-40ea-a4cc-c91b8143f103
 * [new branch]      cron_for_github/459b716e-8e43-4f33-bbc4-843a037228a2 -> origin/cron_for_github/459b716e-8e43-4f33-bbc4-843a037228a2
 = [up to date]      cron_for_github/59304dc6-4c11-4da4-9eb6-fe4885a33d3e -> origin/cron_for_github/59304dc6-4c11-4da4-9eb6-fe4885a33d3e
 = [up to date]      cron_for_github/5e77c64f-5a6a-4686-9028-65279e4ce033 -> origin/cron_for_github/5e77c64f-5a6a-4686-9028-65279e4ce033
 * [new branch]      cron_for_github/5fdf7512-0f07-4039-b89e-5dbf978daee1 -> origin/cron_for_github/5fdf7512-0f07-4039-b89e-5dbf978daee1
 * [new branch]      cron_for_github/600d03f3-06cd-4ae6-969e-ca0a2637b2d3 -> origin/cron_for_github/600d03f3-06cd-4ae6-969e-ca0a2637b2d3
 * [new branch]      cron_for_github/6d33d9f5-6ffb-48ed-8d1c-0beb3f80083a -> origin/cron_for_github/6d33d9f5-6ffb-48ed-8d1c-0beb3f80083a
 * [new branch]      cron_for_github/72108b38-087d-4ecc-bbf8-20f747debac2 -> origin/cron_for_github/72108b38-087d-4ecc-bbf8-20f747debac2
 * [new branch]      cron_for_github/76cf3a60-4f03-48bd-8978-59bfd50dfb57 -> origin/cron_for_github/76cf3a60-4f03-48bd-8978-59bfd50dfb57
 * [new branch]      cron_for_github/84f2ae4f-fa4b-4ba1-82c9-c92d81fe5ccb -> origin/cron_for_github/84f2ae4f-fa4b-4ba1-82c9-c92d81fe5ccb
 * [new branch]      cron_for_github/950cad52-e69e-4755-af27-a588340c600b -> origin/cron_for_github/950cad52-e69e-4755-af27-a588340c600b
 = [up to date]      cron_for_github/a13c68b7-31e7-4ab6-9f7e-a6ec54afc7de -> origin/cron_for_github/a13c68b7-31e7-4ab6-9f7e-a6ec54afc7de
 * [new branch]      cron_for_github/aa14b7c8-8917-42b3-b23d-d3c3a2cfbcd1 -> origin/cron_for_github/aa14b7c8-8917-42b3-b23d-d3c3a2cfbcd1
 * [new branch]      cron_for_github/b01c5c88-00c8-44cf-8057-c6d64ee45ba6 -> origin/cron_for_github/b01c5c88-00c8-44cf-8057-c6d64ee45ba6
 * [new branch]      cron_for_github/b29cae93-09cd-4179-aecd-5b7a160dfdeb -> origin/cron_for_github/b29cae93-09cd-4179-aecd-5b7a160dfdeb
 = [up to date]      cron_for_github/b4526ed5-eae5-4def-af90-522175cac5ff -> origin/cron_for_github/b4526ed5-eae5-4def-af90-522175cac5ff
 * [new branch]      cron_for_github/b8f79728-f366-4c2c-b398-43370577fdf3 -> origin/cron_for_github/b8f79728-f366-4c2c-b398-43370577fdf3
 * [new branch]      cron_for_github/d776f053-365e-499e-976f-c2756af908f2 -> origin/cron_for_github/d776f053-365e-499e-976f-c2756af908f2
 * [new branch]      cron_for_github/d8a68c3a-293e-493a-b834-483d9d7a279f -> origin/cron_for_github/d8a68c3a-293e-493a-b834-483d9d7a279f
 = [up to date]      cron_for_github/db81909c-f728-4614-afea-b26aba69214d -> origin/cron_for_github/db81909c-f728-4614-afea-b26aba69214d
 * [new branch]      cron_for_github/dea383fc-e2c7-454d-ab25-4c2cd5f717a8 -> origin/cron_for_github/dea383fc-e2c7-454d-ab25-4c2cd5f717a8
 = [up to date]      cron_for_github/f96360bf-b400-40f1-91a4-0fc0554f346e -> origin/cron_for_github/f96360bf-b400-40f1-91a4-0fc0554f346e
 = [up to date]      cron_for_github/fe97d780-0642-40a5-b0fe-3ed95ed9428b -> origin/cron_for_github/fe97d780-0642-40a5-b0fe-3ed95ed9428b
 = [up to date]      revert-222-add_robot_init -> origin/revert-222-add_robot_init
 * [new tag]         0.3.10     -> 0.3.10
 * [new tag]         0.3.9      -> 0.3.9
 * [new tag]         0.3.8      -> 0.3.8

```

k-okada, 2017-05-12:
Ok, removed
Can you add cron settings to Travis, so I can remove heroku settings for
this

knorth55, 2017-05-12:
I have no authority to modify travis setting.
Can you authorize me?

knorth55, 2017-05-15:
I confirmed that @k-okada set travis cron build, so close this issue.

closed 2017-05-15 by knorth55