jsk-ros-pkg/jsk_pr2eus #327: Travis CI dependency problem?

Affonso-Gui, 2017-10-31:
Travis CI build ends up with the following error while executing `setup_ros` (#325 #326) . Is this a dependency problem?

```
++ sudo easy_install pip==6.0.7
Searching for pip==6.0.7
Reading http://pypi.python.org/simple/pip/
Couldn't retrieve index page for 'pip'
Scanning index of all packages (this may take a while)
Reading http://pypi.python.org/simple/
No local packages or download links found for pip==6.0.7
error: Could not find suitable distribution for Requirement.parse('pip==6.0.7')
+++ error
```

furushchev, 2017-10-31:
Please wait for https://github.com/jsk-ros-pkg/jsk_travis/pull/349

Affonso-Gui, 2017-10-31:
Ok, thanks!

Closing for being duplicate of jsk-ros-pkg/jsk_travis#349

closed 2017-10-31 by Affonso-Gui