jsk-ros-pkg/jsk_pr2eus #346: Problems running a tutorial

ericlesaquiles, 2018-05-13:
@furushchev  I tried to run tutorials from [https://github.com/furushchev/jsk_pr2eus/blob/f19c477d8bf089947cfdba34fe102b024a070004/pr2eus_tutorials/README.md] by following the instructions on that page (the commands shown as follows). I have tried running them with different local configurations (changing python version, switching to bash back and forth), with the same results. 

obs.: lines starting with "→" mark beginning of command line.

The problem seems to be mostly with finding two packages (namely, ros-kinetic-pr2-moveit-plugins and ros-kinetic-pr2-moveit-config). Why that is, it beats me. The package seems to exist for kinetic (there is [[http://wiki.ros.org/pr2_moveit_plugins](http://wiki.ros.org/pr2_moveit_plugins)], and the jenkins page lists a kinetic version, but I don't know if it is possible to download it from there (I haven't managed to)), but I cannot find a way of downloading it..

```
➜  mal git:(master) rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"
Add distro "hydro"
Add distro "indigo"
Add distro "jade"
Add distro "kinetic"
Add distro "lunar"
Add distro "melodic"
updated cache in /home/ericles/.ros/rosdep/sources.cache
➜  mal git:(master) cd ~ 
backup             kernoops           syslog           
clickpkg           lightdm            systemd-bus-proxy
colord             list               systemd-network  
ericles            proxy              systemd-resolve  
festival           redis              systemd-timesync 
gnats              root               uuidd            
guest-rihxbg       saned              whoopsie         
hplip              speech-dispatcher  www-data         
irc                sys                                 
➜  mal git:(master) cd ~/ros 
➜  ros ls
kinetic  ws_rtmros_tutorials
➜  ros ls kinetic 
build  devel  logs  src
➜  ros sdo rm -r kinetic 
[sudo] password for ericles: 
sem1NSorry, try again.
[sudo] password for ericles: 
➜  ros mkdir -p ~/ros/kinetic/src && cd ~/ros/kinetic/src
➜  src wstool init
Writing /home/ericles/ros/kinetic/src/.rosinstall

update complete.
➜  src his wstool
 1336  source ~/.zshrc
 1337  sudo apt-get install python-rosinstall python-rosinstall-generator python-wstool build-essential
 1338  sdo apt install rospack 
--
 1399  eusgl
 1400  sudo apt-get install python-wstool\n
 1401  sdo apt install python-catkin-tools
--
 1407  cd catkin_ws
 1408  wstool init src
 1409  ls
--
 1417  cd src
 1418  wstool set rtm-ros-robotics/rtmros_tutorials https://github.com/start-jsk/rtmros_tutorials.git --git
 1419  wstool update rtm-ros-robotics/rtmros_tutorials
 1420  cd rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials
--
 1582  ls
 1583  wstool init src
 1584  catkin init
--
 4447  source .zshrc
[... 134 lines cut ...]
;; loading roseus("1.6.3") on euslisp((9.23 ip-172-30-1-231 Thu Feb 22 20:55:14 PST 2018  1.1.0))
;; Writing /home/ericles/ros/kinetic/src/jsk-ros-pkg/jsk_pr2eus/pr2eus/speak.l document to speak.md ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...Done
;; Writing /home/ericles/ros/kinetic/src/jsk-ros-pkg/jsk_pr2eus/pr2eus/robot-interface.l document to robot-interface.md ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...;; skip require (ros::modname ros::dirfile) ...; [... cut ...]
[build] Runtime: 17.0 seconds total.                                                                                                                
[build] Note: Workspace packages have changed, please re-source setup files to use them.
➜  kinetic source ~/.zshrc                             
Talkers are no good doers.
		-- William Shakespeare, "Henry VI"
➜  kinetic rosrun pr2eus_tutorials pr2_tabletop_sim.sh
[rosrun] Couldn't find executable named pr2_tabletop_sim.sh below /opt/ros/kinetic/share/pr2eus_tutorials
➜  kinetic 

```

furushchev, 2018-05-16:
sorry for trouble, pr2 moveit is only available on shadow fixe as of today (2018/5/15)
http://repositories.ros.org/status_page/kinetic_default.html?q=pr2-moveit-config

ericlesaquiles, 2018-05-16:
Thank you, it worked for me

closed 2018-05-16 by ericlesaquiles

ericlesaquiles, 2018-05-23:
By the way that, just to be sure, that exotic inline citation comes from `fortune` :)