jsk-ros-pkg/jsk_pr2eus #348: Missing file from tutorial installation

ericlesaquiles, 2018-05-23:
I have successfully installed the demos on page [[https://github.com/furushchev/jsk_pr2eus/blob/f19c477d8bf089947cfdba34fe102b024a070004/pr2eus_tutorials/README.md](https://github.com/furushchev/jsk_pr2eus/blob/f19c477d8bf089947cfdba34fe102b024a070004/pr2eus_tutorials/README.md)] by making use of shadow-fix.
However, it should be noted that, when running the examples, the file `pr2_tabletop_sim.sh` was missing, so I had to get it separately.

furushchev, 2018-05-23:
Hi,
Which branch are you using?
Since this readme is assumed that the branch will be merged into master branch, it guides to run

```bash
wstool set jsk-ros-pkg/jsk_pr2eus --git https://github.com/jsk-ros-pkg/jsk_pr2eus.git -v master
```

but it is still not the latest one.
 If you run the command above, please run commands below in the source directory in your catkin workspace:

```bash
wstool set jsk-ros-pkg/jsk_pr2eus --git https://github.com/furushchev/jsk_pr2eus.git -v pr2eus-tabletop-test-sample -u -y
```

In this branch there is `pr2_tabletop_sim.sh` and other updates.
Bringing only `pr2_tabletop_sim.sh` from this forked branch into master branch will not work if you are running whole tutorials.

ericlesaquiles, 2018-05-24:
I used the `wstool set jsk-ros-pkg/jsk_pr2eus --git https://github.com/furushchev/jsk_pr2eus.git -v pr2eus-tabletop-test-sample -u -y` command (I just checked). I figured copying only `pr2_tabletop_sim.sh`, so I copied the whole directory (sorry, I didn't express myself correctly). And it at least seemed to work.. sometimes the robot's arm moves so slowly it seems to be still, but I suppose it not to be an unexpected feature. To be sure, I've just ran the installing procedure once more, step by step and it went out really fine (no need to copy the files this time), and the demos are running the same as before (as far as I could see). 

There were some errors during the running, which you may be already aware of, but they didn't affect the demo as far as I could notice. Namely:

(in bright red)
```

ERROR: cannot launch node of type [jsk_interactive_marker/bounding_box_marker]: jsk_interactive_marker
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share
```

(many of the following kind) 
```

TIFFFetchNormalTag: Warning, Incompatible type for "RichTIFFIPTC"; tag ignored.
TIFFFieldWithTag: Internal error, unknown tag 0x829a.
```

Sometimes it also fails to build convex, and stops half way:

```
qhull precision warning:                                                                                                                    [16/1649]
The initial hull is narrow (cosine of min. angle is 0.9999999999999998).
Is the input lower dimensional (e.g., on a plane in 3-d)?  Qhull may
produce a wide facet.  Options 'QbB' (scale to unit box) or 'Qbb' (scale
last coordinate) may remove this warning.  Use 'Pp' to skip this warning.
See 'Limitations' in qh-impre.htm.
qhull precision warning: 
The initial hull is narrow (cosine of min. angle is 0.9999999999999993).
Is the input lower dimensional (e.g., on a plane in 3-d)?  Qhull may
produce a wide facet.  Options 'QbB' (scale to unit box) or 'Qbb' (scale
last coordinate) may remove this warning.  Use 'Pp' to skip this warning.
See 'Limitations' in qh-impre.htm.
qhull precision warning: 
The initial hull is narrow (cosine of min. angle is 0.9999999999999997).
Is the input lower dimensional (e.g., on a plane in 3-d)?  Qhull may
produce a wide facet.  Options 'QbB' (scale to unit box) or 'Qbb' (scale
last coordinate) may remove this warning.  Use 'Pp' to skip this warning.
See 'Limitations' in qh-impre.htm.
qhull precision warning: 
The initial hull is narrow (cosine of min. angle is 0.9999999999999994).
Is the input lower dimensional (e.g., on a plane in 3-d)?  Qhull may
produce a wide facet.  Options 'QbB' (scale to unit box) or 'Qbb' (scale
last coordinate) may remove this warning.  Use 'Pp' to skip this warning.
See 'Limitations' in qh-impre.htm.
[ERROR] [1527125153.297032331, 25.067000000]: failed to build convex
[ERROR] [1527125162.564177160, 26.109000000]: failed to build convex
[ERROR] [1527125171.047191264, 27.087000000]: failed to build convex
[ERROR] [1527125179.696272707, 28.118000000]: failed to build convex
[ERROR] [1527125187.704684957, 29.127000000]: failed to build convex
[ERROR] [1527125194.575093161, 30.118000000]: failed to build convex
[ERROR] [1527125202.368041025, 31.070000000]: failed to build convex
[ERROR] [1527125212.599836036, 32.050000000]: failed to build convex
[ERROR] [1527125223.694720390, 33.115000000]: failed to build convex
[ERROR] [1527125235.537531338, 34.102000000]: failed to build convex
[ERROR] [1527125247.389778870, 35.073000000]: failed to build convex
[ERROR] [1527125254.996712529, 36.082000000]: failed to build convex
[ERROR] [1527125262.444308350, 37.073000000]: failed to build convex
[ERROR] [1527125269.551917815, 38.080000000]: failed to build convex

```
(that one was when trying to grasp the bottle)

And sometimes gazebo or rviz bugs, so I have to delete some log file and run again, or just run again, but, apart from that, I've not had other difficulties with this.

furushchev, 2018-05-24:
```
ERROR: cannot launch node of type [jsk_interactive_marker/bounding_box_marker]: jsk_interactive_marker
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share
```

For this error, please install this package manually: `sudo apt-get install ros-kinetic-jsk-interactive-marker`

And the latter error, just ignore it, since this error occurs only when a robot perceives the too-well-orgnized environment that is unrealistic in the world.

ericlesaquiles, 2018-05-24:
I installed the said package and the first error vanished..
It sounds odd that there is an error for when the environment is too organized, but ok by me

closed 2018-05-24 by ericlesaquiles

k-okada, 2018-05-24:
@furushchev so do we need https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/349 ?

furushchev, 2018-05-24:
The PR is still open so I also added in https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/217.