jsk-ros-pkg/jsk_pr2eus #353: shoulder_pan_joint の effort 値について

weiqiyang, 2018-06-27:
PR2の肩のヨウ回転が追従できなかった実験を再現して、/joint_status のトピックから effort 値を記録しました。
鍋返し動作を5回しました結果、l_shoulder_pan_joint の effort が5回とも11あたりで頭打ちしました(最大値は11.12でした)。
[image]
urdf フィルを見る限り、肩のジョイントの effort は30まで（下記のリンクを参照）のはずですが、トピックに出力された値と urdf ファイルにある値の単位は同じになっていますか。そうであれば、どこからリミッターがかけていますか。
https://github.com/PR2/pr2_common/blob/kinetic-devel/pr2_description/urdf/shoulder_v0/shoulder.urdf.xacro#L26

YoheiKakiuchi, 2018-06-27:
safety_controllerというタグに値が入っていると思うのですが、
http://wiki.ros.org/pr2_controller_manager/safety_limits ここを参考にすると、
速度が出ている時、トルクのリミットが下がるようになっているはずです。
（同じく、速度が出ていると、角度の制限も変化します）
これが働いていそうか、確認してみるとどうでしょうか？

止まった状態で、外から力をかけて30Nmまでeffortが出るかも確認できるかと思います。

weiqiyang, 2018-06-27:
サーボが入ったまま、外力でゆっくり動いたり、力をかけたまま静止したりしながら、
またデータをとってみましたが、effort はやはり+-１１くらいまでしか出られなかった。
（最小値-11.09、最大値11.11）

YoheiKakiuchi, 2018-06-27:
https://github.com/PR2/pr2_robot/blob/indigo-devel/pr2_controller_configuration/pr2_arm_controllers.yaml#L13
ここの値がそれらしいですね。
shoulder_liftもリミットになっているように見えていて、それも同じファイルに書いてある値に制限されていそうです。

urdfの記述と、コントローラの記述のそれぞれがどういう由来の値かを考える必要があるように思います。

weiqiyang, 2018-06-27:
urdfとコントローラのファイルをひと通り見ましたが、
urdfファイルの方は、腕10、グリッパー1000、その他30と一見適当な数値になっています；
コントローラにはshoulder_pan 11.8、shoulder_lift 8.0、それ以外はリミットを設定していません。

前述の11.8のリミットは
https://github.com/PR2/pr2_robot/commit/5283af265531056c4f73f39f2317b9812e719b0c#diff-612c4f0f682717066eacca312e98a01f
のコミットで追加されていましたが、その理由については不明ゆえ、
@furushchev さんに編集者へ質問のメールを書いてもらいました。

個人的には、PR2安全講習ビデオのとあるシーンが衝撃でありましたため、
挟み込み事故の防止など、安全面の配慮じゃないかと。

k-okada, 2018-06-28:
> PR2/pr2_robot@5283af2#diff-612c4f0f682717066eacca312e98a01f

大変良い調査ですね．で，次はこういうふうになる，ということが，ロボットを動かす前に予想できるかどうかしりたいです．

１） この動作を作るプログラムで生成される，関節の軌道列をプロットしてみてください．十分に指令値が密だったらOK,疎だったら
https://github.com/euslisp/jskeus/blob/1.1.0/irteus/test/interpolator.l#L27
で密な軌道をつくってみてください．
２） この時の，速度，加速度が実機のjonit_stateで観測されたものと同等だったか調べてみてください．
３）
この時の各関節にかかるトルクを以下の方法で計算してみてください．これと実機で観測されたものを比較してみてください．よりよいトルクの計算方法がないか，@mmurooka
等に聞いてみてください．
https://github.com/euslisp/jskeus/blob/1.1.0/irteus/irtrobot.l#L384

これで，実機と同じような情報が取れたら，例えば速度をかえて，どのぐらいの速度だとeffortのリミットにかからないか，あるいは，円運動の円の大きさや向きを変えた時の違いを評価することができるようになるとおもいます．それで実機の制約以下になるものを選んで，そのとおりに動くかどうか，知りたいです．

mmurooka, 2018-07-02:
> この時の各関節にかかるトルクを以下の方法で計算してみてください．これと実機で観測されたものを比較してみてください．よりよいトルクの計算方法がないか，@mmurooka

遅くなりました．以下で動的な慣性も考慮したトルクを計算できます．分からなければまた直接聞いてください．

```
(load "package://pr2eus/pr2-interface.l")

(setq *robot* (instance pr2-robot :init))
(objects (list *robot*))

(setq i 0)
(do-until-key
 (incf i)
 ;; 1. set joint-angle
 (send (send *robot* :joint "r_shoulder_lift_joint") :joint-angle (* 40 (sin i)) :relative t)
 ;; 2. calculate torque (dtは前回のangle-vectorから今回のangle-vectorまでの間の時間 [sec])
 (send *robot* :calc-torque :calc-statics-p t :dt 0.01)
 ;; 3. get joint torque (単位は[Nm])
 (print (send-all (send *robot* :joint-list) :joint-torque))
 ;; 4. draw
 (send *irtviewer* :draw-objects)
 (send *robot* :draw-torque (send *irtviewer* :viewer) :flush t :size 200.0 :width 4)
 )
```
[image]

weiqiyang, 2018-07-04:
皆様、ありがとうございます。
編集者から返信をもらいましたので、その内容を伝えします。
PR2の肩のモーターが腕全体の動作に関連している割に、出力は非常に弱く(very week)、激しい動作の実現は難しいとのことです。
よって、エフォートに関しては11.8の方が正確な数値だそうです。
PR2自体は頑丈なので、リミットを外しても壊れないらしい。

YoheiKakiuchi, 2018-07-04:
メールの文面をどこか研究室内で共有できるところに置いてもらえるかな。
研究室外に公開してもよさそうなら、ここに貼ってくれてもいい。

weiqiyang, 2018-07-04:
編集者本人がメールの公開には反対していますゆえ、その旨で前のコメントを編集させました。
同じ理由で文面についてはこちらで貼ることはありませんが、後ほど @furushchev さんと相談して、可能であれば内部だけで共有するようにします。

closed 2018-07-04 by weiqiyang

k-okada, 2018-09-14:
@weiqiyang BTW, any update on this project?

weiqiyang, 2018-09-19:
> @weiqiyang BTW, any update on this project?

Not really. I am working on something else recently.