jsk-ros-pkg/jsk_pr2eus #354: end-coords-interpolation cannot take less than 10s

pazeshun, 2018-07-09:
```
(send *robot* :rarm :move-end-pos #f(0 0 300) :world)
(send *ri* :angle-vector (send *robot* :angle-vector) 1000 nil 0 :end-coords-interpolation t)
```
does not move robot in 1 second, but 10 seconds.

Affonso-Gui, 2018-07-09:
try playing around with `:end-coords-interpolation-steps` and `:min-time`

pazeshun, 2018-07-09:
`:angle-vector`'s default `min-time` is 1 sec.
This is applied to interpolated steps of `end-coords-interpolation`.
`end-coords-interpolation-steps`'s default value is 10, so execution time becomes 1 * 10 = 10 secs.

@Affonso-Gui Thank you for your comment. I already found solution. See #355

pazeshun, 2018-07-09:
> :angle-vector's default min-time is 1 sec.
> This is applied to interpolated steps of end-coords-interpolation.
> end-coords-interpolation-steps's default value is 10, so execution time becomes 1 * 10 = 10 secs.

Maybe this is confusing behavior because usual `:angle-vector`'s `min-time` is minimum time for whole execution.

PR #355 (2018-09-21): Equalize min-time behavior of end-coords-interpolation to usual angle-vector
Fixes #354 
Includes #378 

cc @Affonso-Gui
closed 2018-09-21 by k-okada