jsk-ros-pkg/jsk_pr2eus #397: closing right gripper fails

AnneKoepken, 2019-10-01:
Getting error message when executing the below code snippet, joint configuration is not allowed. The problem appears in the last line, when trying to close the right gripper.
 @Affonso-Gui 

(load "package://pr2eus/pr2-interface.l")
(pr2)
(setq *ri* (instance pr2-interface :init))

(send *pr2* :r_shoulder_lift_joint :joint-angle -20)
(send *pr2* :r_upper_arm_roll_joint :joint-angle -130)
(send *pr2* :r_elbow_flex_joint :joint-angle -80)
(send *pr2* :r_forearm_roll_joint :joint-angle 0)
(send *pr2* :r_wrist_flex_joint :joint-angle -70)
(send *ri* :angle-vector (send *pr2* :angle-vector) 3000)
(send *ri* :wait-interpolation)

(send *ri* :start-grasp :rarm :gain 0.05) ;;this line does not work

Affonso-Gui, 2019-10-01:
What I got until now:

- Minimal code:
```lisp
(load "package://pr2eus/pr2-interface.l")
(pr2-init)

(send *pr2* :reset-pose)
(send *pr2* :rarm :angle-vector #f(-60.0 -20.0 -130.0 -80.0 0.0 -70.0 180.0))
(send *ri* :angle-vector (send *pr2* :angle-vector) 3000)
(send *ri* :wait-interpolation)

(send *ri* :start-grasp :rarm)
```
- The problem happens in `convex-hull-3d` function, in the last body class initialization https://github.com/euslisp/EusLisp/blob/master/lisp/geo/primt.l#L430-L434 with the error message `float vector expected`

- It is sensitive enough that if I dump the vector list with 3 decimal points precision and load it again the error does not occur

- The function is being called in :start-grasp to check for collision with any given objects (which, in this case, do not exist). Therefore, it is unnecessary calculation in this (and many other) case(s).

**Conclusion**:
Avoid the unnecessary calculation on :start-grasp until we figure out the real problem in `convex-hull-3d`?
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/pr2-interface.l#L270-L277

knorth55, 2019-10-01:
Good catch! we can skip this process when `objects` is `nil`.

Affonso-Gui, 2019-10-01:
Created PR at https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/398

knorth55, 2019-11-02:
@Affonso-Gui 
Can you check if this issues is solve by  #398 ?

Affonso-Gui, 2019-11-05:
Checked the code, running fine.
@AnneKoepken Can we close this?

closed 2019-11-05 by k-okada