jsk-ros-pkg/jsk_pr2eus #433: kinematics simulatorでのassoc等がよくわからない

k-okada, 2020-04-21:
From @Kanazawanaoaki 

やかんやマグカップなどを持ち上げる動作などを作っているのですが、上手く腕についていかず、物体だけ先に瞬間移動して次の動画のようになってしまいます。

[image]

IRTviewer上とKinematics simulator上の両方でassocされるように現在は、
(send *pr2* :larm :end-coords :assoc *kettle*)
(send *ri* :objects (send *room73b2* :objects))
(send *pr2* :larm :end-coords :assoc *kettle*)
のようにしています。

仕様なのかもしれませんし、実機を動かす上では問題なさそうなので、このままでも良いですが、良い解決策をご存知の方がいたら教えていただけるとありがたいです。

k-okada, 2020-04-21:
kinematics simulator ないでの物体の扱いは
https://gist.github.com/k-okada/eec8654d068240ab7ff9cbaad12d2bd9
という使い方が想定されます．見るべきコードは
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/pr2-interface.l#L265-L293
になります．

現状ではpr2-interface に実装されているので，他のロボットで使いたい場合はrobot-interfaceにうつしていく必要があると思います．

k-okada, 2020-04-21:
ちなみに，kinematics simulator の中の物品は`body` 単一か，複数のbody がassoc されているものは，`bodyset-link` 担っていないとダメ？

[image]

Kanazawanaoaki, 2020-04-27:
start-graspでつかめない複雑な形状の物体については
```lisp
(send *ri* :robot :rarm :end-coords :assoc (car (send *ri* :find-object *kettle-lid*)))
```
のようにして無理やりassocしました。

[image]

pazeshun, 2020-05-01:
@k-okada 
ひとまず #435 を作って、グリッパ内の物体とのassoc/dissocを、PR2以外のロボットからも呼べるようにしました。
ただ、各ロボットで`:start-grasp`や`:stop-grasp`を作らないといけない状況は変わっていないので、これで良いのかは謎です。