jsk-ros-pkg/jsk_pr2eus #436: add scrennpoint example

k-okada, 2020-05-12:
> topicを使ってPointcloudScreenpointを呼ぶためのサンプルプログラムについてもプルリクエストを出したので、参考になれば嬉しいです。
> https://github.com/MiyabiTane/rtmros_tutorials/pull/2

これ，めちゃめちゃ便利＆必要だと思うので，https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus_tutorials/launch/pr2_tabletop_grasp.launch でimage_view2をクリックしたら，その位置をgraspするモードも作ってくれると嬉しいです．

708yamaguchi, 2020-05-12:
image_view2を使ったインタラクティブなgraspモードを実装してみました。
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/437

knorth55, 2020-05-12:
コメントが遅くなりましたがこのサンプルでは元々Rvizのバウンディングボックスを選択するとPR2が掴むデモになっています
そちらを壊さないように実装してほしいです

>

knorth55, 2020-05-12:
勘違いしていました。
reviewでもかきましたがargumentで分けるのではなく、常にrvizでもimage view2でもできるようにどちらも起動すればいいとおもいます。
全て一つでできるlaunchのほう`pr2_tabletop_grasp.launch`のほうを使うようにオススメするのであれば、今のreadmeのeus, rvizを順番で自分の手で立ち上げる項目は消した方がわかりやすいと思います。
個人的な趣味で、動作実行、可視化を同じlaunchで立ち上げるのが不便（gazeboや実機が突然動き出す）ので別にしていました。
あと普通に実機のほうでは画像処理ノードをPR2体内で、eus, rvizはローカルPCで立ち上げることが想定されているので、`pr2_tabpletop_grasp.launch`の使いどころはないと思います。

708yamaguchi, 2020-05-12:
ありがとうございます。
以下の方針でプルリクエストを修正します。

- pr2-tabletop-object-grasp.l 
  - `Rviz`からでも`image_view2`からでも把持物体位置を受け取れるようにする（今のまま）
- pr2_tabletop.launch
  - 常に`Rviz`も`image_view2`も使えるようにする
- pr2_tabletop_sim.launch
  - gazeboの立ち上げ + pr2_tabletop.launch
- pr2_tabletop_grasp.launch
  - deperecatedなので、変更しない。
- README
  - pr2_tabletop_grasp.launchは使わないようにする。

closed 2020-05-15 by k-okada