jsk-ros-pkg/jsk_pr2eus #474: Error in inverse-kinematics fail with :move-target

Affonso-Gui, 2022-01-27:
I still need to make a deeper search on the exact cause for this problem, but I have had problems in the warning callback on :inverse-kinematics with `:move-target`.

How to replicate:
```lisp
(send *pr2* :larm :inverse-kinematics (make-coords) :move-target (send *pr2* :l_wrist_flex_link_lk))
```

Error when compiled from source:
```lisp
;; inverse-kinematics failed.
;; dif-pos : #f(-88.3792 17.7853 -390.194)/(400.472/10)
;; dif-rot : #f(0.132979 0.01128 -0.001116)/(0.133461/0.087266)
;;  coords : #<coordinates #X55ff8d40cd28  0.0 0.0 0.0 / 0.0 0.0 0.0>
;;  angles : (299.833 51.3541 25.7066 103.755 -102.215 3.49524 -15.107 -90.0015 -60.0002 -12.0727 -78.1486 -84.58 -79.6035 -101.302 -633.226 0.426701 73.926)
;;    args : ((#<coordinates #X55ff908347f8  0.0 0.0 0.0 / 0.0 0.0 0.0>) :move-target #<bodyset-link #X55ff8c860060 l_wrist_flex_link  83.963 34.267 390.069 / 0.002 -0.011 -0.133> :link-list (#<bodyset-link #X55ff8c87d8c0 l_shoulder_pan_link  -50.0 188.0 1090.508 / -0.565 0.0 0.0> #<bodyset-link #X55ff8c87dcb0 l_shoulder_lift_link  34.48 134.492 1090.508 / -0.565 1.296 -1.110e-16> #<bodyset-link #X55ff8c87dde8 l_upper_arm_roll_link  34.48 134.492 1090.508 / -0.565 1.296 2.606> #<bodyset-link #X55ff8c881c40 l_elbow_flex_link  126.078 76.476 705.483 / -2.355 1.384 0.841> #<bodyset-link #X55ff8c881e38 l_forearm_roll_link  126.078 76.476 705.483 / -2.355 1.384 -2.347> #<bodyset-link #X55ff8c860060 l_wrist_flex_link  83.963 34.267 390.069 / 0.002 -0.011 -0.133> #<bodyset-link #X55ff8c86a8c0 l_wrist_roll_link  83.963 34.267 390.069 / 0.002 -0.011 3.045>) :rthre 0.087266 :thre 10 :stop 300 :additional-weight-list ((#<bodyset-link #X55ff8d5bd0a8 torso_lift_link  -50.0 0.0 1090.508 / 0.0 0.0 0.0> 0.005)) :link-list (#<bodyset-link #X55ff8c87d8c0 l_shoulder_pan_link  -50.0 188.0 1090.508 / -0.565 0.0 0.0> #<bodyset-link #X55ff8c87dcb0 l_shoulder_lift_link  34.48 134.492 1090.508 / -0.565 1.296 -1.110e-16> #<bodyset-link #X55ff8c87dde8 l_upper_arm_roll_link  34.48 134.492 1090.508 / -0.565 1.296 2.606> #<bodyset-link #X55ff8c881c40 l_elbow_flex_link  126.078 76.476 705.483 / -2.355 1.384 0.841> #<bodyset-link #X55ff8c881e38 l_forearm_roll_link  126.078 76.476 705.483 / -2.355 1.384 -2.347> #<bodyset-link #X55ff8c860060 l_wrist_flex_link  83.963 34.267 390.069 / 0.002 -0.011 -0.133> #<bodyset-link #X55ff8c86a8c0 l_wrist_roll_link  83.963 34.267 390.069 / 0.002 -0.011 3.045>) :move-target #<bodyset-link #X55ff8c860060 l_wrist_flex_link  83.963 34.267 390.069 / 0.002 -0.011 -0.133> :move-target #<bodyset-link #X55ff8c860060 l_wrist_flex_link  83.963 34.267 390.069 / 0.002 -0.011 -0.133> :collision-avoidance-link-pair nil :link-list (#<bodyset-link #X55ff8c87d8c0 l_shoulder_pan_link  -50.0 188.0 1090.508 / -0.565 0.0 0.0> #<bodyset-link #X55ff8c87dcb0 l_shoulder_lift_link  34.48 134.492 1090.508 / -0.565 1.296 -1.110e-16> #<bodyset-link #X55ff8c87dde8 l_upper_arm_roll_link  34.48 134.492 1090.508 / -0.565 1.296 2.606> #<bodyset-link #X55ff8c881c40 l_elbow_flex_link  126.078 76.476 705.483 / -2.355 1.384 0.841> #<bodyset-link #X55ff8c881e38 l_forearm_roll_link  126.078 76.476 705.483 / -2.355 1.384 -2.347> #<bodyset-link #X55ff8c860060 l_wrist_flex_link  83.963 34.267 390.069 / 0.002 -0.011 -0.133> #<bodyset-link #X55ff8c86a8c0 l_wrist_roll_link  83.963 34.267 390.069 / 0.002 -0.011 3.045>) :move-target #<bodyset-link #X55ff8c860060 l_wrist_flex_link  83.963 34.267 390.069 / 0.002 -0.011 -0.133>)
Call Stack (max depth 20):
  0: at (apply #'send-message self (class . super) :inverse-kinematics target-coords :rthre rthre :thre thre :stop stop :additional-weight-list (list (list (send self :torso_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005))) :link-list link-list :move-target move-target args)
  1: at (apply #'send-message self (class . super) :inverse-kinematics target-coords :rthre rthre :thre thre :stop stop :additional-weight-list (list (list (send self :torso_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005))) :link-list link-list :move-target move-target args)
  2: at (apply #'send-message self (class . super) :inverse-kinematics target-coords :rthre rthre :thre thre :stop stop :additional-weight-list (list (list (send self :torso_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005))) :link-list link-list :move-target move-target args)
  3: at (apply #'send-message self (class . super) :inverse-kinematics target-coords :rthre rthre :thre thre :stop stop :additional-weight-list (list (list (send self :torso_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005))) :link-list link-list :move-target move-target args)
  4: at (apply #'send-message self (class . super) :inverse-kinematics target-coords :rthre rthre :thre thre :stop stop :additional-weight-list (list (list (send self :torso_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005))) :link-list link-list :move-target move-target args)
  5: at (apply #'send-message self (class . super) :inverse-kinematics target-coords :rthre rthre :thre thre :stop stop :additional-weight-list (list (list (send self :torso_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005))) :link-list link-list :move-target move-target args)
  6: at (send-super* :inverse-kinematics target-coords :rthre rthre :thre thre :stop stop :additional-weight-list (list (list (send self :torso_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005))) :link-list link-list :move-target move-target args)
  7: at (cond (use-base (setq diff-pos-rot (concatenate float-vector (send start-coords :difference-position self) (send start-coords :difference-rotation self))) (send self :move-to start-coords :world) (with-append-root-joint (ll self link-list :joint-class omniwheel-joint :joint-args base-range) (send (caar ll) :joint :joint-angle (float-vector (elt diff-pos-rot 0) (elt diff-pos-rot 1) (rad2deg (elt diff-pos-rot 5)))) (send-super* :inverse-kinematics target-coords :rthre rthre :thre thre :stop stop :additional-weight-list (append (list (list (send self :torso_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005)) (list (car (send self :links)) (if (eq use-base t) 0.1 use-base))) additional-weight-list) :link-list ll :move-target move-target args))) (t (send-super* :inverse-kinematics target-coords :rthre rthre :thre thre :stop stop :additional-weight-list (list (list (send self :torso_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005))) :link-list link-list :move-target move-target args)))
  8: at (let (diff-pos-rot) (unless move-arm (setq move-arm (send self :select-target-arm target-coords))) (unless move-target (if (consp move-arm) (setq move-target (mapcar #'(lambda (arm) (send self arm :end-coords)) move-arm)) (setq move-target (send self move-arm :end-coords)))) (unless link-list (setq link-list (if (consp target-coords) (mapcar #'(lambda (target) (let ((l target) move-arm) (while l (cond ((memq l (send self :larm)) (setq move-arm :larm)) ((memq l (send self :rarm)) (setq move-arm :rarm))) (setq l (send l :parent))) (send self :link-list (send target :parent) (unless use-torso (car (send self move-arm)))))) move-target) (send self :link-list (send move-target :parent) (unless use-torso (car (send self move-arm))))))) (cond (use-base (setq diff-pos-rot (concatenate float-vector (send start-coords :difference-position self) (send start-coords :difference-rotation self))) (send self :move-to start-coords :world) (with-append-root-joint (ll self link-list :joint-class omniwheel-joint :joint-args base-range) (send (caar ll) :joint :joint-angle (float-vector (elt diff-pos-rot 0) (elt diff-pos-rot 1) (rad2deg (elt diff-pos-rot 5)))) (send-super* :inverse-kinematics target-coords :rthre rthre :thre thre :stop stop :additional-weight-list (append (list (list (send self :torso_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005)) (list (car (send self :links)) (if (eq use-base t) 0.1 use-base))) additional-weight-list) :link-list ll :move-target move-target args))) (t (send-super* :inverse-kinematics target-coords :rthre rthre :thre thre :stop stop :additional-weight-list (list (list (send self :torso_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005))) :link-list link-list :move-target move-target args))))
  9: at (send *pr2* :larm :inverse-kinematics (make-coords) :move-target (send *pr2* :l_wrist_flex_link_lk))
[... cut ...] o_lift_joint :child-link) (if (numberp use-torso) use-torso 0.005))) :link-list link-list :move-target move-target args))))
  9: at (double_stereo_link_lk . #<bodyset-link #X564c795d8df8 double_stereo_link  69.328 1.016 1467.524 / 0.007 1.29 -8.674e-19>)
  10: at diff-pos-rot
  11: at ;p=pointer?(0x564c742e51e8)
*** buffer overflow detected ***: /opt/ros/melodic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl terminated
Aborted (core dumped)
```

Setting `:warnp` to nil seems to avoid the problem.

Naoki-Hiraoka, 2022-01-27:
This is not pr2eus's bug. This is jskeus's bug.

```lisp
(load "package://euslisp/jskeus/irteus/demo/sample-robot-model.l")
(setq *sr* (instance sample-robot :init))
(send *sr* :rarm :inverse-kinematics (make-coords) :move-target (send *sr* :link :rleg-link5))
```

```
; inverse-kinematics failed.
;; dif-pos : #f(0.0 75.0 650.0)/(654.313/1)
;; dif-rot : #f(0.0 0.0 0.0)/(0.0/0.017453)
;;  coords : #<coordinates #X5641921bb6b0  0.0 0.0 0.0 / 0.0 0.0 0.0>
;;  angles : (0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0)
;;    args : ((#<coordinates #X5641921d40e8  0.0 0.0 0.0 / 0.0 0.0 0.0>) :move-target #<bodyset-link #X564191489d70 :rleg-link5  0.0 -75.0 -650.0 / 0.0 0.0 0.0> :link-list (#<bodyset-link #X564192272838 :rarm-link0  0.0 -150.0 175.0 / 3.142 0.524 3.142> #<bodyset-link #X5641922a0d40 :rarm-link1  0.0 -150.0 175.0 / 3.142 0.524 -3.111> #<bodyset-link #X5641922db5d0 :rarm-link2  0.0 -150.0 175.0 / 1.556 0.026 2.618> #<bodyset-link #X564192316928 :rarm-link3  -137.437 -141.671 413.048 / -0.138 1.043 1.451> #<bodyset-link #X5641923342c0 :rarm-link4  -130.685 52.86 401.352 / -0.0 -0.52 1.502> #<bodyset-link #X564192375168 :rarm-link5  -130.685 52.86 401.352 / -0.0 -0.52 -0.069> #<bodyset-link #X5641910f6238 :rarm-link6  -130.685 52.86 401.352 / 0.067 -1.037 -0.118>) :move-target #<bodyset-link #X564191489d70 :rleg-link5  0.0 -75.0 -650.0 / 0.0 0.0 0.0> :collision-avoidance-link-pair nil :link-list (#<bodyset-link #X564192272838 :rarm-link0  0.0 -150.0 175.0 / 3.142 0.524 3.142> #<bodyset-link #X5641922a0d40 :rarm-link1  0.0 -150.0 175.0 / 3.142 0.524 -3.111> #<bodyset-link #X5641922db5d0 :rarm-link2  0.0 -150.0 175.0 / 1.556 0.026 2.618> #<bodyset-link #X564192316928 :rarm-link3  -137.437 -141.671 413.048 / -0.138 1.043 1.451> #<bodyset-link #X5641923342c0 :rarm-link4  -130.685 52.86 401.352 / -0.0 -0.52 1.502> #<bodyset-link #X564192375168 :rarm-link5  -130.685 52.86 401.352 / -0.0 -0.52 -0.069> #<bodyset-link #X5641910f6238 :rarm-link6  -130.685 52.86 401.352 / 0.067 -1.037 -0.118>) :move-target #<bodyset-link #X564191489d70 :rleg-link5  0.0 -75.0 -650.0 / 0.0 0.0 0.0>)
Call Stack (max depth: 20):
  0: at (send *sr* :rarm :inverse-kinematics (make-coords) :move-target (send *sr* :link :rleg-link5))
  1: at (send *sr* :rarm :inverse-kinematics (make-coords) :move-target (send *sr* :link :rleg-link5))
  2: at (send *sr* :rarm :inverse-kinematics (make-coords) :move-target (send *sr* :link :rleg-link5))
  3: at (send *sr* :rarm :inverse-kinematics (make-coords) :move-target (send *sr* :link :rleg-link5))
  4: at (send *sr* :rarm :inverse-kinematics (make-coords) :move-target (send *sr* :link :rleg-link5))
  5: at (send *sr* :rarm :inverse-kinematics (make-coords) :move-target (send *sr* :link :rleg-link5))
  6: at #<compiled-code #X564191192e18>
/opt/ros/melodic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :parent in (send *sr* :rarm :inverse-kinematics (make-coords) :move-target (send *sr* :link :rleg-link5))
```

原因は、デバッグメッセージを表示するときに、
move-targetが(end-coordsのようなロボットのリンクに含まれないcascoordsではなく)ロボットのリンクの場合、デバッグメッセージが読みやすくなるように
`#<bodyset-link #X555c69b3b808 :rleg-link5  0.0 -75.0 -650.0 / 0.0 0.0 0.0>`から`(send r :link :rleg-link5)`へと置き換えるのですが
https://github.com/euslisp/jskeus/blob/317c29c67d52648af485c4833a7fbfd6d8345415/irteus/irtmodel.l#L2578
置き換えたものに対して`cascaded-coords`型が想定された処理が行われているためです。
https://github.com/euslisp/jskeus/blob/317c29c67d52648af485c4833a7fbfd6d8345415/irteus/irtmodel.l#L2581-L2586

Affonso-Gui, 2022-01-27:
@Naoki-Hiraoka 

Thanks for looking it up! Am closing this in favor of https://github.com/euslisp/jskeus/pull/611

closed 2022-01-27 by Affonso-Gui